ophub/amlogic-s9xxx-openwrt #274: 江苏版M401A：U盘启动正常、安装到EMMC后无法启动，另大量kmod无法更新

xcray, 2022-07-21:
openwrt_s905l3a_k5.15.55_2022.07.16.0420 用U盘启动一切正常，约23秒即可启动完毕，非常nice！

尝试安装到emmc后，安装过程正常，关机拔掉U盘再重新加电，发现无法启动：
[image]
只好继续用U盘。。。

另外，发现大量kmod是5.10.127版本的，提示有5.10.131版本可以升级，可是我用的是5.15.55的内核，所以升级失败。。。等5.15.55的模块？
[image]
感谢ophub大佬，即使U盘启动，也足以使得这个小盒子重获新生！

ophub, 2022-07-21:
这些模块如果你需要，可以在编译固件的时候添加在.config里面编译。

xcray, 2022-07-21:
> 这些模块如果你需要，可以在编译固件的时候添加在.config里面编译。例如：
> .......
> 后期就忽略吧

实话说我想把这些模块都删掉。。。应该都没啥用吧？与内核版本不一致，是不是想用也只能重新编译呀？

另外，还是请大佬分析一下为啥写到emmc里就启动不了呢？

ophub, 2022-07-21:
忽略，不用删。你看一下https://docs.qq.com/doc/DWFprWWxZa290VXNk

xcray, 2022-07-21:
> 忽略，不用删。你看一下https://docs.qq.com/doc/DWFprWWxZa290VXNk

这个文档看了好几遍，有收获也有被带歪的情况，汗。。。

看文档里所说，似乎openwrt写emmc是一个无解的问题？我先用U盘跑着吧，反正emmc也嫌不够大呢