ophub/amlogic-s9xxx-openwrt #275: 切换内核失败：Error：The UBOOT_OVERLOAD file is missing and cannot be update.

xcray, 2022-07-24:
江苏M401A，S905L3A，openwrt_s905l3a_k5.15.55_2022.07.22.0713.img.gz
根据文档说明切换内核版本，将 boot-xxx.tar.gz, dtb-amlogic-xxx.tar.gz, modules-xxx.tar.gz 这 3 个内核文件上传完成后界面确实显示了 更新内核 的操作按钮，点击后提示：
The UBOOT_OVERLOAD file is missing and cannot be update.

没搞懂该怎么整。。。

xcray, 2022-07-24:
删除3个内核文件，尝试手动上传完整版固件，点操作按钮后同样失败：The device name is empty and cannot be recognized.

xcray, 2022-07-24:
尝试在线下载更新，也不成功：02.02 Invalid firmware check.

ophub, 2022-07-24:
你的固件在哪下载的？
`ls /boot`
截图

xcray, 2022-07-24:
> 你的固件在哪下载的？ `ls /boot` 截图

就在这个仓库的Release里下的。刚看了一下，/boot是空的，启动之后没有挂载。

手动挂载之后，上传失败：
[image]
提交的安全令牌无效或已过期！

ophub, 2022-07-24:
s905l3a的openwrt不支持写入emmc使用，你在usb中使用，启动后执行openwrt-tf命令会一键分成4个区

xcray, 2022-07-25:
重新烧了一遍。。。

closed 2022-07-25 by xcray