ophub/amlogic-s9xxx-openwrt #284: 支持玩客云S805 CPU

mysmartha, 2022-08-14:
armbian官方已经支持玩客云了，玩客云armbian Linux内核已经支持5.15、5.18了，但是网上能找到的玩客云openwrt固件Linux内核还是老旧的3.x，玩客云价格低廉、运行稳定，有千兆网口，非常适合刷openwrt做旁路由，希望大佬能出手支持一下，造福众多玩客云玩家，谢谢！

ophub, 2022-08-15:
发一下你测试可以使用的玩客云的armbian固件下载地址，以及刷机方法

mysmartha, 2022-08-15:
固件下载地址 https://cloud.189.cn/t/QfM7VjrQZVZv (访问码: o7cb) 里面有线刷包和usb启动刷写包，我用的是USB刷写包。
刷入方法：
1、电脑连接玩客云靠近hdmi的USB口，短接刷机触点或者按住复位键，使用Aml Burn Tool软件烧录底包至玩客云。
烧录完成后断开玩客云电源备用。
2、解压下载的固件（可能需要解压多次，直至解压出.img文件），使用USBWriter软件将固件写入U盘。写入成功后，先将U盘插入玩客云靠近网线接口端的USB口，再接入电源。 玩客云通电后指示灯会先红粉闪烁，稍等5-10分钟会红黄规律性闪烁，等到最后灯光熄灭后，再拔掉U盘、电源。重新上电，刷机结束。

初始账号密码  root   1234
该包应该是作者参考https://github.com/hzyitc/armbian-onecloud 制作的，请大佬参考

mysmartha, 2022-08-15:
这里有固件和源码https://github.com/hzyitc/armbian-onecloud/releases

ophub, 2022-08-16:
https://github.com/coolsnowwolf/lede/commit/f6c1512c108db56d681d28c539e0dc745deedf23

lean前几天己经支持了s805的编译

closed 2022-08-17 by ophub

jwkinko, 2022-10-08:
> [酷雪狼/lede@ f6c1512](https://github.com/coolsnowwolf/lede/commit/f6c1512c108db56d681d28c539e0dc745deedf23)
> 
> 精益前几天己经支持了s805的编译

不支持写入emmc