ophub/amlogic-s9xxx-openwrt #285: github action parameters not working

0x5e, 2022-08-18:
Hi,
I fork the repo and trigger the `Build OpenWrt with lede`, and I found the parameters are not working...
I see them in logs are still default value, but actually I'm choosing `s905l3a` and `5.4.210_5.10.135_5.15.50`:
```
2022-08-18T06:21:41.4768386Z ##[group]Run ophub/amlogic-s9xxx-openwrt@main
2022-08-18T06:21:41.4768653Z with:
2022-08-18T06:21:41.4768995Z   openwrt_path: openwrt/bin/targets/*/*/*rootfs.tar.gz
2022-08-18T06:21:41.4769230Z   openwrt_soc: all
2022-08-18T06:21:41.4769431Z   openwrt_kernel: 5.10.135_5.15.50
2022-08-18T06:21:41.4769635Z   auto_kernel: true
2022-08-18T06:21:41.4769809Z   openwrt_size: 960
2022-08-18T06:21:41.4769981Z   mode: ophub
2022-08-18T06:21:41.4770166Z   version_branch: stable
```

ophub, 2022-08-18:
You have set invalid parameters, delete:
```
mode: ophub
version_branch: stable
```

0x5e, 2022-08-18:
the code in [build-openwrt-with-lede.yml](https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/.github/workflows/build-openwrt-with-lede.yml#L139) is variable ${{ github.event.inputs.xxx}}, no `mode: ophub` exists:
```
      - name: Build OpenWrt firmware
        if: steps.compile.outputs.status == 'success' && !cancelled()
        uses: ophub/amlogic-s9xxx-openwrt@main
        with:
          openwrt_path: openwrt/bin/targets/*/*/*rootfs.tar.gz
          openwrt_soc: ${{ github.event.inputs.openwrt_soc }}
          openwrt_kernel: ${{ github.event.inputs.openwrt_kernel }}
          auto_kernel: ${{ github.event.inputs.auto_kernel }}
          openwrt_size: ${{ github.event.inputs.openwrt_size }}

```

ophub, 2022-08-18:
[image]

ophub, 2022-08-18:
https://github.com/ophub/amlogic-s9xxx-openwrt/runs/7897294460?check_suite_focus=true

0x5e, 2022-08-18:
I start the workflow like this, but now I'm not 100% sure now, I can't see the input in the log :(
[image]

ophub, 2022-08-18:
https://github.com/ophub/amlogic-s9xxx-openwrt/runs/7897734209?check_suite_focus=true

[image]
[image]

0x5e, 2022-08-18:
Sorry, my mistake

closed 2022-08-18 by 0x5e