ophub/amlogic-s9xxx-openwrt #286: CM311-1A 写入emmc 无法启动

yingzaigt, 2022-08-19:
大佬好，我的CM311-1A USB可以正常启动您2022.8.17发布的s905l3a  openwrt，但是用晶晨宝盒写入emmc后无法启动；CM311-1A 刷您的armbian 用armbian-install no写入emmc，USB引导启动一次后正常emmc启动（armbian-install 写入emmc无法启动）。所以想问一下openwrt有没有还是用安卓原来bootloader，不写主线bootloader的方法呀？感谢大佬维护的armbian 和openwrt ，造福我等爱折腾的玩家。

ophub, 2022-08-19:
1. 先还原安卓TV系统
2. 把这个文件 `/etc/model_database.txt` 里面的 `/lib/u-boot/e900v22c-u-boot.bin.sd.bin` 改为 `NA`
3. 再使用宝盒安装，这样就去掉这个主线U-BOOT了，很多人反馈这个主线使用后无法启动，我下午把他先去掉吧

closed 2022-08-20 by ophub

ysuolmai, 2022-09-27:
我用最新版的还是一样问题，[openwrt_s905l3a_k5.15.70_2022.09.24.img.gz]，U盘可以启动，写入emmc就无法启动了。还需要改/etc/model_database.txt吗？

ophub, 2022-09-27:
不用改了，选择305
你写入armbian可以，写入openwrt不行？

ysuolmai, 2022-09-27:
> 不用改了，选择305
> 你写入armbian可以，写入openwrt不行？

对，armbian完全没问题了. op写进emmc就起不来了，而且这时候u盘也没法引导了

ophub, 2022-09-27:
你使用的是什么时候的固件？
op的安装文件里面，分区的大小要改一下，少cm311的划分

hhalibo, 2022-09-29:
> 你使用的是什么时候的固件？ op的安装文件里面，分区的大小要改一下，少cm311的划分

5天前的新版本，也是刷入emmc不能启动。请问大佬，需要在哪里改分区大小，请明示，谢谢！

ophub, 2022-09-29:
https://github.com/ophub/luci-app-amlogic/blob/d125217b6ee47ff3516cfbdd3018a5647c0d71ff/luci-app-amlogic/root/usr/sbin/openwrt-install-amlogic#L329

329行前添加：
```
elif [[ "${AMLOGIC_SOC}" == "s905l3a" ]]; then
    BOOT=512
    BLANK1=108
    BLANK2=778
    BLANK3=0
    BLANK4=0
```

hhalibo, 2022-09-29:
> https://github.com/ophub/luci-app-amlogic/blob/d125217b6ee47ff3516cfbdd3018a5647c0d71ff/luci-app-amlogic/root/usr/sbin/openwrt-install-amlogic#L329
> 
> 329行前添加：
> 
> ```
> elif [[ "${AMLOGIC_SOC}" == "s905l3a" ]]; then
>     BOOT=512
>     BLANK1=108
>     BLANK2=778
>     BLANK3=0
>     BLANK4=0
> ```

我去试试，非常感谢！

hhalibo, 2022-09-29:
报告大佬，刷入eMMC启动成功！非常感谢！

ophub, 2022-09-29:
好的，恭喜你

CC91300, 2025-08-15:
> 报告大佬，刷入eMMC启动成功！非常感谢！

您好，能分享下固件包吗