ophub/amlogic-s9xxx-openwrt #288: X96 mini - Python2

deyloTT, 2022-08-24:
Hey, so i have a python script that i usually run with "python2 script.py" because python3 throws errors. I just got openwrt running on a x96 mini android box and for some reason i cant get python2 installed (i got python3 to install however). Can i be enlightened as to how i can get python2 running? Thanks

ophub, 2022-08-24:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/9d9d27f78e416d2245d04cd9506958d86d438eef/router-config/openwrt-master/.config#L3143

[image]

From the official source code, support for python2 has been removed

deyloTT, 2022-08-24:
Oh boy :( And there isn’t anway to 'sideload' it through community package or something like that?

ophub, 2022-08-24:
This is the official configuration file of github.com/openwrt, only supports python3

deyloTT, 2022-08-24:
Ive found a repository which has the packages that gets added successfully but i get error:
Packages for python-light found, but incompatible with the architectures configured

ophub, 2022-08-24:
What you are looking for is x86_64, the box is `aarch64`

deyloTT, 2022-08-24:
https://downloads.openwrt.org/releases/packages-18.06/aarch64_generic/packages/

thanks. would this architecture work? Im seeing the packages there also.

ophub, 2022-08-24:
ok

deyloTT, 2022-08-24:
> What you are looking for is x86_64, the box is `aarch64`

on my opkg package list configuration, i see that im receiving packages in "/armvirt/64/packages"
is this correct or did i use an incorrect release? Im using s905l x96 mini

closed 2022-08-24 by deyloTT

ophub, 2022-08-25:
The software package of aarch64 can be used in amlogic s9xxx devices, regardless of soc

deyloTT, 2022-08-25:
Ok thanks...one more thing off-tolic: is it possible to install openvpn on my device? Im able to install openvpn-rsa but not the actual openvpn server.

ophub, 2022-08-25:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/9d9d27f78e416d2245d04cd9506958d86d438eef/router-config/lede-master/.config#L537-L538

It is included in the `lede` version, but not in the official version. You need to install version 21.03 yourself.

https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_Aml_lede_08.21.2259

deyloTT, 2022-08-25:
Aahhh i see. Haha. Thanks. Ill have to flash over the box with the lede version then. Thanks for the help

deyloTT, 2022-08-25:
I just installed the lede version and its in chinese. Anyway to switch to english? Or do i have to download another version? I downloaded the one in the link you sent above.