ophub/amlogic-s9xxx-openwrt #289: TX 8MAX openwrt 无线无法启动

Fengke6, 2022-08-24:
O大我又来测试TX 8MAX openwrt了，无线无法启动，网卡不能识别。
[image]
另外armbian下可以搜到蓝牙，无法连接，我认为这个dtb,是不完美的。

ophub, 2022-08-24:
你的驱动己经正常识別了，剩下的就是测试你的无线设置，比如国家改us，通道分别试试，比如n1选择36就可以正常使用。

ophub, 2022-08-24:
armbian 里面，先在armbian-config下安装蓝牙管理模块

Fengke6, 2022-08-24:
> 你的驱动己经正常识別了，剩下的就是测试你的无线设置，比如国家改us，通道分别试试，比如n1选择36就可以正常使用。

根据你的提示测试结果如下:
1. 5G ap模式无法发射信号，更换时区和信道没有作用
2.  2.4g ap可以发射信号，手机无法获取ip。

ophub, 2022-08-24:
你是什么模式，没ip是dhcp的事，如果是主路由模式，把lan和无线桥接选择上。如果是旁路由模式，看你的主路由怎么设置的。

简单的测试方法是把盒子和家里的网络路由器断开。把盒子的lan和无线桥接选择上。把lan的dhcp打开（不选择取消dhcp），把lan里面的ip/网关都写openwrt自己的静态ip，手机电脑连盒子的wifi，能访问盒子的ip打开openwrt即可，因为没上网所以其他网站打不开。

Fengke6, 2022-08-24:
> armbian 里面，先在armbian-config下安装蓝牙管理模块

下载了的

ophub, 2022-08-24:
2.4/5G一般都是同时只有一个可用，没给接入的设备分ip是dhcp没设置好。

盒子的无线和硬路由ap功能比很弱，带机量/稳定性/覆盖范围都不好。这是一个电视盒子，他本身不是路由器，无线芯片的性能只够自己联网用，一般来说都不用

ophub, 2022-08-24:
盒子装openwrt使用，一般是为了使用op丰富的插件功能，以旁路由的方式接入交换机，补充传统硬路由没有的很多插件的功能。无线建议用硬路由的。

Fengke6, 2022-08-24:
> 盒子装openwrt使用，一般是为了使用op丰富的插件功能，以旁路由的方式接入交换机，补充传统硬路由没有的很多插件的功能。无线建议用硬路由的。

是的我用的是旁路由，无线是硬路由，我只是测试下无线网卡

closed 2022-08-25 by ophub