ophub/amlogic-s9xxx-openwrt #292: wetek model 4kiptvh01 wont boot black screen

murad0522, 2022-09-06:
hello  i have wetek device   
it have 905x + 2GB ram + 8 rom + 100MB ethernet + ap6255 

I tried most of dtb files inside but no boot, 

how I can extract the file , or know the right one , 

the device running android 9 

thank you

ophub, 2022-09-06:
try: https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_Aml_lede_09.06.0653/openwrt_s905x_k5.15.62_2022.09.06.img.gz

murad0522, 2022-09-08:
same thing, no boot. do I need to specify the dtb file?

ophub, 2022-09-08:
Do you have a product link? Motherboard high-definition picture, network card chip picture

No code output at all? Did you enable USB boot mode?

murad0522, 2022-09-08:
i dont knwo to enable  USB boot mode I use sd card only 

motherboard high-definition picture, network card chip picture , I can take picture

ophub, 2022-09-08:
that is not supported

murad0522, 2022-09-08:
no product link from wetek ,

those links for the motherboard 

https://ibb.co/kgK0smX
https://ibb.co/kBdDxb8
https://ibb.co/cc4QXjx
https://ibb.co/s58g73F
https://ibb.co/64t5Gr5

ophub, 2022-09-08:
[image]

[image]

ophub, 2022-09-08:
Looking at the photo, it is possible to boot with the default firmware of the s905x. Did you turn on USB debugger mode?

murad0522, 2022-09-08:
yes i did.. but when i connect usb cable nothing happen  no new device

ophub, 2022-09-09:
If your device has a reset button, after you try to power on, press and hold for 5 seconds to see if it can enter the flash mode

murad0522, 2022-09-09:
after long time of trying i figure out how to boot from usb... it's turned out sd boot not working only usb disk. and i should wait for 10 sec and screen will flash two times then it  start... but when i tried to install the openwrt to internal storage its wont boot eeven if i choose same dtb file it's booted from usb.. it's end with bootloop

ophub, 2022-09-09:
which version of the kernel are you using

murad0522, 2022-09-09:
i tried this img openwrt_s905x_k5.10.138_2022.08.26.img
also have a problem with the network  as there is no internet most of the time, I put lan to static IP and config DNS in lan and disable LAN  DHCP because I have another DHCP server running when I connect through wifi the phone is connected and all goes well expect no internet  most of the time and if there is internet it's limited I mean there is website wont open like speedtest

ophub, 2022-09-09:
try 5.15

murad0522, 2022-09-09:
update, i installed it to internal storage by using this kernel 5.15, but the network problem still exists I hope someone helps me how to  solve it, I tried to use dynamic IP  but still same problem, I need to switch from lan to wan, but I don't know how to do it and if I delete lan interface wan would work

ophub, 2022-09-09:
Installed. How to use Internet query: openwrt use

closed 2022-09-09 by ophub