ophub/amlogic-s9xxx-openwrt #299: S905l3A-B 使用 Kernel 5.15 lan口 无反应

WenMoshi, 2022-09-19:
您好，设备是s905l3a-b的e900v22c，在使用 Kernel 5.10.143时设备可以正常驱动，e900v22c的网口用网线连接路由器或电脑都正常工作且可以访问后台无问题。但在使用Kernel 5.15.68时设备似乎无法驱动，网线连接路由器或电脑都没有反应(确定不是网线问题)，当然没有连通也就无法进入后台。前几个版本无论是Official还是lede都可以复现此问题，不知道问题出现在哪里，看看大神能不能解决这个问题。谢谢ophub大神！膜拜！

WenMoshi, 2022-09-19:
根据您的方法，设备已经正确驱动并进入后台无问题，问题得到解决，再次感谢ophub大神！

closed 2022-09-19 by ophub

WenMoshi, 2022-09-19:
似乎新问题又出现了，引导的内核是5.10.143而不是5.15.68，在系统-概览-内核显示是5.15.68，但是软件包里的kernel是5.10.143-1-a9330f9fc9..b4a，而且软件包里关于kmod的相关几乎全部都是5.10.143-1。

WenMoshi, 2022-09-19:
好的，非常感谢，我在研究研究。[笑脸]

WenMoshi, 2022-09-25:
op大神，是否能将软件包5.10.143编译为官方snapshots软件包版本(目前是5.15.69)，添加ipk相关依赖极其痛苦。尤其是kmod相关依赖，内核不匹配让我感到绝望[doge]。

ophub, 2022-09-25:
kmod要在编译时添加，你fork仓库，自己添加更多个性化软件包。添加方法在仓库的说明文档里。

WenMoshi, 2022-09-27:
明白您的意思，现阶段正在启蒙学习，从零学习很艰苦现在还不会编译，如果您能将releases里发布的镜像软件包kernel及kmod相关编译为5.15，那就对新手小白太友好了，因为添加依赖我就会从这里找(https://downloads.openwrt.org/snapshots/targets/armvirt/64/packages/或https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/)，所以就产生了上述问题。望大神能理解！

ophub, 2022-09-27:
有两个release，一个是lean的，一个是openwrt.org官方的，kmod是使用的他们源码默认的版本。

WenMoshi, 2022-10-09:
> > op大神，是否能将软件包5.10.143编译为官方snapshots软件包版本(目前是5.15.69)，添加ipk相关依赖极其痛苦。尤其是kmod相关依赖，内核不匹配让我感到绝望[doge]。
> 
> 我遇到过类似问题，我在 https://mirrors.cloud.tencent.com/lede/snapshots/targets/armvirt/64/packages 和 https://mirrors.cloud.tencent.com/lede/snapshots/targets/armvirt/64/kmods 这里手动下载kernel，luci-web中手动上传、安装kernel。 再 使用单独命令 opkg install libgcc1 opkg install libstdcpp6 opkg install libatomic1 opkg install libpthread opkg install librt 接着 opkg list-upgradable | cut -f 1 -d ' ' | xargs opkg upgrade
> 
> e.g. opkg ###################5.15.71 src/gz openwrt_core https://mirrors.cloud.tencent.com/lede/snapshots/targets/armvirt/64/packages src/gz openwrt_base https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_cortex-a53/base src/gz openwrt_kmods http://mirrors.cloud.tencent.com/lede/snapshots/targets/armvirt/64/kmods/5.15.71-1-bdf35995d529c12d697be46e1592ce8a src/gz openwrt_luci https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_cortex-a53/luci src/gz openwrt_packages https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_cortex-a53/packages src/gz openwrt_routing https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_cortex-a53/routing src/gz openwrt_telephony https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_cortex-a53/telephony
> 
> 另外，我发现报内核依赖有问题完全可以忽略。貌似除了passW，其他都能正常安装、使用

好的，感谢您分享的方法，我试试