ophub/amlogic-s9xxx-openwrt #304: More a question rather than an issue - build ext4 instead of btrfs?

Falcordog, 2022-09-30:
I used provided GH Actions to build OpenWRT using the ImageBuilder method.
The compiled images are nice and clean which is great!
However btrfs is always causing read-only issues - at least on the images i made.
admittedly I have little experience with btrfs and that could may well be the issue.
Tried to extroot-overlay, combining both internal EMMC and external SD but thtat always seemed to break things.
btrfs is very protective of its snapshot functionality which always ended up locking up my ROOTFS to the point where I couldn't use opkg anymore.
Hence my query - is it possible to build ext4 images (ImageBuilder), and possibly extrooting the internal ext4 to external f2fs devices (USB,SD). 
Or even having a choice to run OpenWRT on external storage alone using f2fs.
I had my NeWiFI D2 exrooted to a f2fs USB3 flash drive which worked very well until i shifted thingds around.
That said, the EMMC on the router used squashfs (not ext4) which never caused any read-only lockoups and I'd much rather have that as an alternative instead of btrfs too.
Thanks!

ophub, 2022-09-30:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/ba845d627a32d304a73c24cc6786b5e7f1cfe417/make#L673

https://github.com/ophub/luci-app-amlogic/blob/7cdd63d7c1ff49ee87ccd68ef7b791cac0121306/luci-app-amlogic/root/usr/sbin/openwrt-install-amlogic#L478

https://github.com/ophub/luci-app-amlogic/blob/7cdd63d7c1ff49ee87ccd68ef7b791cac0121306/luci-app-amlogic/root/usr/sbin/openwrt-install-amlogic#L485

closed 2022-09-30 by ophub

remlei, 2022-10-04:
nice, just in case someone is lost there, those links above pertains on a command what filesystem the partition formats, change the btrfs to ext4 should do the trick but there are other stuff that needs to be done as well or else the if you run the openwrt-install-amlogic script will just complain about cant mount stuff (since other parts of the script still assumes btrfs, and there some adjustments needs to be done as well).

either way I posted a modified script below, that basically handles all of the stuff.  make sure you booted to USB install of openwrt and run that script. Doesnt matter if the USB install is still under btrfs filesystem, as it just run a cp command anyway when copying the rootfs content on the USB. 

[openwrt-install-amlogic-ext4.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/9701989/openwrt-install-amlogic-ext4.zip)

protip: you can also adjust ROOT1=960 to something smaller,  since my openwrt-imagbuilder rootfs.bz is 210mb big, I basically set it to ROOT1=300MB