ophub/amlogic-s9xxx-openwrt #306: T95Z PLUS - Controlling the LED display dtb

creamisty, 2022-10-02:
hello. can you add openvfd in meson-gxm-t95z-plus.dtb for next build?
I have tried using the configuration from https://github.com/arthur-liberman/vfd-configurations/blob/master/t95z-1-vfd.conf, and it works fine for me. thanks

[image]

ophub, 2022-10-02:
You can already use it, it means that it has been added in dtb.

creamisty, 2022-10-02:
for wired network and wireless network works fine on my 95z plus.
wireless network using bcrm-AP6255, nand sandisk, ram 3/32. 

[image]
[image]
[image]
[image]

LED work using your meson-gxm-t95z-plus-2.dtb and config from t95z-1-vfd.conf.

========================

[image]

tried modified value from your share, make LED is off on me.

ophub, 2022-10-03:
According to your feedback, gigabit wired, wireless, led can work fine. I will submit this dtb file to the upstream kernel repository. I'll add the led's configuration file to the list of configuration files.

ophub, 2022-10-03:
According to user feedback, t95z must be started using `/boot/extlinux/extlinux.conf`, which file are you using? Did you rename extlinux.conf.bak to extlinux.conf to start?

I see you are using `uEnv.txt` when testing dtb?

creamisty, 2022-10-03:
> According to your feedback, gigabit wired, wireless, led can work fine. I will submit this dtb file to the upstream kernel repository. I'll add the led's configuration file to the list of configuration files.

OK. thanks.

> According to user feedback, t95z must be started using `/boot/extlinux/extlinux.conf`, which file are you using? Did you rename extlinux.conf.bak to extlinux.conf to start?
> 
> I see you are using `uEnv.txt` when testing dtb?

yes, im used file extlinux.conf.bak. oh oke. I forgot to rename the file to extlinux.conf. I will try it

creamisty, 2022-10-03:
the result : 

1. config from: https://github.com/arthur-liberman/vfd-configurations/blob/master/t95z-1-vfd.conf 

LED = ON
```
vfd_gpio_clk='0,71,0'
vfd_gpio_dat='0,72,0'
vfd_gpio_stb='0,70,0'

vfd_chars='0,1,2,3,4'
vfd_dot_bits='0,1,2,3,4,5,6'

vfd_display_type='0x01,0x00,0x00,0x00'
```
[image]

2.  config from your modified value : 

LED = OFF
```
vfd_gpio_clk='0,70,0'
vfd_gpio_dat='0,71,0'
vfd_gpio_stb='0,69,0'

vfd_chars='0,1,2,3,4'
vfd_dot_bits='0,1,2,3,4,5,6'

vfd_display_type='0x01,0x00,0x00,0x00'
```
3. config from : https://github.com/arthur-liberman/vfd-configurations/blob/master/t95z-2-vfd.conf

LED = OFF
```
vfd_gpio_clk='0,68,0'
vfd_gpio_dat='0,67,0'
vfd_gpio_stb='0,69,0'

vfd_chars='4,3,2,1,0'
vfd_dot_bits='0,1,2,3,4,5,6'

vfd_display_type='0x01,0x00,0x00,0x00'
```

4. modified value from config NO. 3 : 

LED = ON , sometimes symbol wifi or lan or etc is on or off
```
vfd_gpio_clk='0,71,0'
vfd_gpio_dat='0,72,0'
vfd_gpio_stb='0,70,0'

vfd_chars='4,3,2,1,0'
vfd_dot_bits='0,1,2,3,4,5,6'

vfd_display_type='0x01,0x00,0x00,0x00'
```
My time = HH:MM = 14:49
[image]

My time = HH:MM = 14:50
[image]

creamisty, 2022-10-03:
Is there a difference when I use android stock firmware from 7.1.2 and android 9?

ophub, 2022-10-03:
```
vfd_gpio_clk='0,67,0'
vfd_gpio_dat='0,66,0'
vfd_gpio_stb='0,68,0'

vfd_chars='4,3,2,1,0'

vfd_dot_bits='0,1,2,3,4,5,6'

vfd_display_type='0x01,0x00,0x00,0x00'

functions='wifi eth usb alarm play pause'
```

test this result

creamisty, 2022-10-03:
> ```
> vfd_gpio_clk='0,67,0'
> vfd_gpio_dat='0,66,0'
> vfd_gpio_stb='0,68,0'
> 
> vfd_chars='4,3,2,1,0'
> 
> vfd_dot_bits='0,1,2,3,4,5,6'
> 
> vfd_display_type='0x01,0x00,0x00,0x00'
> 
> functions='wifi eth usb alarm play pause'
> ```
> 
> test this result

LED = OFF

ophub, 2022-10-03:
```
vfd_gpio_clk='0,71,0'
vfd_gpio_dat='0,72,0'
vfd_gpio_stb='0,70,0'

vfd_chars='0,1,2,3,4'

vfd_dot_bits='0,1,2,3,4,5,6'

vfd_display_type='0x01,0x00,0x00,0x00'

functions='wifi eth usb alarm play pause'
```
try again

creamisty, 2022-10-03:
> ```
> vfd_gpio_clk='0,71,0'
> vfd_gpio_dat='0,72,0'
> vfd_gpio_stb='0,70,0'
> 
> vfd_chars='0,1,2,3,4'
> 
> vfd_dot_bits='0,1,2,3,4,5,6'
> 
> vfd_display_type='0x01,0x00,0x00,0x00'
> 
> functions='wifi eth usb alarm play pause'
> ```
> 
> try again

[image]

works great!!!!

ophub, 2022-10-03:
ok, congratulations

closed 2022-10-03 by ophub