ophub/amlogic-s9xxx-openwrt #310: kernel version incompatible

livelier, 2022-10-14:
OpenWrt_Aml_official_10.13.0329
OpenWrt_Aml_lede_10.13.0352
其中 s905x3-b 固件 测试了 5.10.147 official 版和 lede 都是存在这个问题，
安装软件包是提示
安装 sftp 软件包，可以正常安装
但是安装 bluez 提示内核不兼容

```
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Package openssh-sftp-server (9.0p1-1) installed in root is up to date.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.72-1-bdf35995d529c12d697be46e1592ce8a) for kmod-crypto-hash
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.72-1-bdf35995d529c12d697be46e1592ce8a) for kmod-crypto-null
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.72-1-bdf35995d529c12d697be46e1592ce8a) for kmod-crypto-aead
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.72-1-bdf35995d529c12d697be46e1592ce8a) for kmod-crypto-manager
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.72-1-bdf35995d529c12d697be46e1592ce8a) for kmod-crypto-authenc
```
```
The installed version of package kernel is not compatible, require 5.15.72-1-bdf35995d529c12d697be46e1592ce8a while 5.15.72-1-e6cf3e70c574040b140b70a9066268e4 is installed.
```

另外 official 版 Material 主题 在安装软件包 按钮按下弹出的对话框中，最下边 
`Allow overwriting conflicting package files` 此选项文字是白色，在默认主题下 背景也是白色，造成文字内容显示不出。

[image]

ophub, 2022-10-14:
kmod不能后期安装，需要在编译时添加。
luci插件可以后期安装。

livelier, 2022-10-14:
好的，下午我测试一下4个版本，emmc写入情况。

至于蓝牙和无线驱动以及软件包，按需编译使用吧。我就不测试了。

ophub, 2022-10-14:
openwrt一般不使用蓝牙，wifi模块可以在网络下的无线栏目里看看，有就说明驱动了。

ophub, 2022-10-15:
在openwrt里的这个分配你测试可行吧？

```
elif [[ "${FDTFILE}" == "meson-sm1-skyworth-lb2004-a4091.dtb" ]]; then
    BOOT=512
    BLANK1=108
    BLANK2=562
    BLANK3=0
    BLANK4=0
```

108+512+562=1182
你提交给f大的时候把安装脚本里的第324行这样修改下吧。

livelier, 2022-10-15:
[model_database.txt#L65](https://github.com/ophub/amlogic-s9xxx-openwrt/blob/ab0bff468eb5ce9e6e77d80e41261d6aa553be7b/amlogic-s9xxx/common-files/rootfs/etc/model_database.txt#L65) 修改主线 uboot 路径

`/usr/lib/u-boot/ 修改为 /lib/u-boot/`

[openwrt-install-amlogic#L163](https://github.com/ophub/luci-app-amlogic/blob/8afedeb17ac0f1d05d5bafdbca506df2482c7bac/luci-app-amlogic/root/usr/sbin/openwrt-install-amlogic#L163) 加入主线 uboot 路径

`5f:skyworthbox:s905x3:meson-sm1-skyworth-lb2004-a4091.dtb:u-boot-skyworth-lb2004.bin:/lib/u-boot/skyworth-lb2004-u-boot.bin.sd.bin:NA:4C@2016Mhz,4GB Mem,1Gb Nic,mtk7661rsn`

[openwrt-install-amlogic#L324](https://github.com/ophub/luci-app-amlogic/blob/8afedeb17ac0f1d05d5bafdbca506df2482c7bac/luci-app-amlogic/root/usr/sbin/openwrt-install-amlogic#L324) 修改分区表
```
elif [[ "${AMLOGIC_SOC}" == "s905x3" ]] && [[ "${FDTFILE}" == "meson-sm1-skyworth-lb2004-a4091.dtb" ]]; then
    BOOT=512
    BLANK1=108
    BLANK2=562
    BLANK3=0
    BLANK4=0
```

[image]

没有找到原因，或者我操作错误，4个固件版本，使用晶晨宝盒写入 emmc 后，主线 uboot 都没有被写入 emmc，
但手动写入主线 uboot 可以正常使用。4个版本的无线都不可用，没有被驱动。

使用安卓 bootloader 时，5.15.y 内核需要增加 /boot/u-boot.ext 才可以 usb 启动，否则不能 usb 启动。

ophub, 2022-10-15:
没有写入主线u-boot是路径或者文件名称不对吧？存在即可写入。
把脚本改一下，只判断dtb名即可

ophub, 2022-10-15:
你给f大提交的时候，改以下几处：
https://github.com/unifreq/openwrt_packit/blob/master/files/openwrt-install-amlogic#L324

324行：

```
elif [[ "${FDTFILE}" == "meson-sm1-skyworth-lb2004-a4091.dtb" ]]; then
    BOOT=512
    BLANK1=108
    BLANK2=562
    BLANK3=0
    BLANK4=0
```

163行：

`5f:skyworthbox:s905x3:meson-sm1-skyworth-lb2004-a4091.dtb:u-boot-skyworth-lb2004.bin:/lib/u-boot/skyworth-lb2004-u-boot.bin.sd.bin:NA:4C@2016Mhz,4GB Mem,1Gb Nic,mtk7661rsn`

[amlogic_model_database.txt](https://github.com/unifreq/openwrt_packit/blob/b733f44cdb6b6f2562ce082e03d309b5d7df2467/files/amlogic_model_database.txt#L53)
第53行，和openwrt-install-amlogic的第324行一样即可。

ophub, 2022-10-15:
有主线内核就不需要/boot/u-boot.emmc了(存在也不影响使用)，这个是配合安卓的bootloader 使用的。

ophub, 2022-10-17:
我今天提交内核的时候，把这个安装文件对极光3的修改也提交到f大的打包仓库了。你的u-boot测试好了提交到他的仓库吧。

livelier, 2022-10-17:
我去复刻一下，提交给f大。
o大，建议把 official，lede 固件压缩包 增加一下区分，现在的压缩包下载后，这两个版本容易混淆 。

ophub, 2022-10-17:
你是说文件命名？

livelier, 2022-10-17:
是的，文件名。我在测试的时候，有好几次都刷错版本了 哈哈哈哈

ophub, 2022-10-17:
好的，一会更新下

livelier, 2022-10-17:
像代码一样漂亮，🎉🎉🎉

livelier, 2022-10-17:
已经提交 f大 仓库。
这一版 u-boot 也支持 6.0.y ，但仅用更新内核方式测试。

ophub, 2022-10-17:
好的，感谢你的支持。
有空时把你开发u-boot的方法，调试方法写个说明，教教大家怎么制作。

livelier, 2022-10-17:
好的，感谢o大。
制作这版 u-boot ，是学习[12.11 如何制作 u-boot 文件](https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/armbian-docs/README.cn.md#1211-%E5%A6%82%E4%BD%95%E5%88%B6%E4%BD%9C-u-boot-%E6%96%87%E4%BB%B6)，并借鉴 f大 的源码，以及你的指导。我仅仅是依葫芦画瓢。😂😂😂

常常看到你说学习笔记，可以分享一下，这个记笔记的平台吗？我也记录记录。😂😂😂

ophub, 2022-10-17:
我就是手机自带的备忘录，经常复制保存，反复看。
我们老师说，念书学习这事情，理解不了就先背会，我就是先背会再慢慢消化。

closed 2022-10-21 by ophub