ophub/amlogic-s9xxx-openwrt #315: lan插入无反应，hdmi输出黑屏

Peterzhangnoise, 2022-10-22:
E900V22D，S905L3A-B
使用5.15内核时，lan插入无反应
使用5.10内核时，hdmi黑屏，无法启动

Peterzhangnoise, 2022-10-22:
5.15 在使用这个方法后，hdmi黑屏，无法启动
https://github.com/ophub/amlogic-s9xxx-openwrt/issues/299#issuecomment-1250920231

Peterzhangnoise, 2022-10-23:
我尝试编译5.10.143，无论是lede的还是openwrt原版的rootfs，均无法启动

ophub, 2022-10-23:
https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_Aml_lede_10.21.0614

s905l3a两个内核的都会不能启动？什么也别改保持默认

Peterzhangnoise, 2022-10-24:
> https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_Aml_lede_10.21.0614
> 
> s905l3a两个内核的都会不能启动？什么也别改保持默认

在这个分支情况下，5.15可以启动，5.10不能启动，但是5.15有个问题就是lan不识别，插电脑上灯都不亮那种，回安卓系统试了下盒子的网卡是好的，电脑的网卡也是好的

ophub, 2022-10-24:
meson-gxl-s905l3b-m302a.dtb

用5.15.y的固件试试这个dtb

closed 2022-10-27 by ophub

ophub, 2022-10-27:
5.15用usb启动后，复制下boot下的u-boot试试

Peterzhangnoise, 2022-10-29:
大佬，无解了，你说的所有方法都试了一遍，不行，要么黑屏不启动，要么网卡识别不到

Peterzhangnoise, 2022-10-29:
meson8b-dwmac ff3f0000.ethernet eth0:validation of rmii with support 0000000,00000000,00006280 and advertisement 0000000,00000000,00006280 failed: -22
meson8b-dwmac ff3f0000.ethernet eth0: no phy at addr -1

ophub, 2022-10-29:
看来是不支持