ophub/amlogic-s9xxx-openwrt #316: CM311-1A 写入emmc后，重启时USB硬盘必须手动插拔一次，才能挂载。

popalanluck, 2022-10-24:
大佬，CM311-1A（s905l3a）盒子刷openwrt_lede_s905l3a_k5.15.74_2022.10.21，顺利写入EMMC并运行。

但遇到的问题是，因为USB比EMMC先启动，导致每次开机都会检测盒子上插的USB硬盘，要等很久才能跳过USB，进入EMMC里面的Openwrt。而且，此时USB硬盘灯灭，必须手动插拔一次，才能正常挂载USB硬盘，非常麻烦。

请问，该如何修改UBOOT启动顺序，让EMMC优先于USB启动呢？

ophub, 2022-10-25:
如果emmc优先于usb，那么emmc系统坏了后，系统无法从usb优先启动，怎么重装系统呢？
op默认的boot分区是fat32格式，标签lable=BOOT
你的移动硬盘也是这个格式和标签名称？

按照系统的启动设置，应该不会把非lable=BOOT的其他硬盘当做启动盘。
你可以使用UUID来挂载你的移动硬盘。固定写如fstab。

97668589, 2022-10-25:
大佬怎么安装PassWall，求教

popalanluck, 2022-10-25:
刷openwrt后，试过这样写入fstab：  UUID=7c1d177f-ade1-6b63-a2ed-a1a180df2c15   /mnt/sda1/   ext4   defaults 0 0 
但是，在移动硬盘灯灭的情况下，根本不加载，必须重新插拔上电，才能挂载硬盘。

同一个盒子，刷安卓系统，开机不检测USB，直接先从EMMC启动，然后能正常挂载移动硬盘。但刷成OPenwrt或者armbian以后，开机就会疯狂检测USB，浪费了大量时间。直到超时才跳到EMMC启动，这时移动硬盘已经断电灭灯，无法挂载。

另一个HK1盒子，刷OP以后，把移动硬盘插在USB3.0上，开机也能直接从EMMC启动，再正常挂载。

这种盒子优先从EMMC启动的话，就算EMMC里面的系统坏了，也能通过短接线刷的方式，用Aml_Burn_Tool刷机，重装系统没问题。

所以，才想调整盒子的启动顺序，就像flippy大大发的那个给rock5b的bootloader: 修改了启动顺序：tf->emmc->usb->nvme 。

ophub, 2022-10-25:
他的那个顺序是改u-boot实现的。rock5b 的顺序是在radxa官方u-boot里定的：
include/configs/rockchip-common.h

你学习学习折腾吧

[image]

popalanluck, 2022-10-26:
感谢大佬的解答，自己改了boot.cmd和boot.scr，又换了其它品牌的硬盘盒，问题基本解决。

原来用maiwo的硬盘盒和迅雷送的同品牌易驱线，硬盘盒开机疯狂检测USB硬盘；用易驱线开机不检测，但几十秒过后也疯狂重复检测，造成移动硬盘无法使用。

换了其它品牌的硬盘盒，更改文件后，移动硬盘插在cm311盒子的USB2接口，把openwrt刷入EMMC，现在开机或者重启，基本都能成功挂载。

尝试上传替换回原BOOT文件，也成功实现了用U盘重刷Openwrt。