ophub/amlogic-s9xxx-openwrt #318: /etc/profile.d/30-sysinfo.sh:15: = not found

karnadii, 2022-10-25:
I got this error when login from terminal. 
`/etc/profile.d/30-sysinfo.sh:15: = not found`
my repo https://github.com/karnadii/KarnadiWrt
[image]

it is fine when running `/etc/profile.d/30-sysinfo.sh` directly. but not when login from terminal. this is not `ophub/amlogic-s9xxx-openwrt/` issue, but more my custom build issue. but maybe you know something? I don't know where the error come from. thank you.

ophub, 2022-10-25:
https://github.com/karnadii/KarnadiWrt/blob/b3b1495b5fcb2bfb633e03c2f88be268eaef2b09/make#L562-L563

karnadii, 2022-10-25:
so this is the problem? https://github.com/karnadii/KarnadiWrt/blob/main/router-config/immortalwrt-master/diy-part2.sh#L47
thanks, will try to do something about it and test again.

closed 2022-10-25 by karnadii

karnadii, 2022-10-25:
fix it, thanks.