ophub/amlogic-s9xxx-openwrt #323: 能否在luci里面增加adguard插件

Edwardyx, 2022-10-30:
luci里自带的广告大师+感觉不给力。然后就发现可以docker adguard，结果第一次拉取的时候失败了。我就网上搜，说是换个docker源，按照步骤换了，然后说是要重启docker `systemctl restart docker`之类的，记不怎么住。然后就卡死了。突然想起来docker还在跑openwrt，然后就发现上不了网。等了半天也不行。
我就重新开机。
### 然后`armbian-software`就发现了了不得的神奇的事情！！！除了docker，potainer和openwrt全都变成未安装状态了！

也不知道是不是自动被删了。装个过滤器都这么难的吗。
我就想着，要没大家都没有，你个docker凭什么独活，删了重装。
重装docker以后发现没有运行，嗯也不知道怎么然它运行，就reboot。
### 然后了不得的神奇的事情叒发生了！！！ssh 登录不了！！！
我太难了
第一次：周五晚上6点半搞到1点多+周六上午9开始搞到下午3点，第一次OK
第二次：周六想要全家电视手机都屏蔽广告，就想openwrt作为DHCP，结果㕛开始从下午2点到晚上11点，第二次OK
第三次：嗯，1、2、3，开始。。。
**祝大佬周末愉快T_T**

ophub, 2022-10-30:
adguard的源码仓库地址发一下

Edwardyx, 2022-10-30:
我找到了这个：

[luci-app-adguardhome](https://github.com/kongfl888/luci-app-adguardhome)
Adguard不知道具体是哪个：
[https://github.com/AdguardTeam](https://github.com/AdguardTeam)

ophub, 2022-10-31:
.config 里加 
CONFIG_PACKAGE_luci-app-adguardhome=y

diy2.sh 里加
svn co https://github.com/rufengsuixing/luci-app-adguardhome/trunk package/luci-app-adguardhome

closed 2022-10-31 by ophub