ophub/amlogic-s9xxx-openwrt #333: x96max plus 2101w网口异常

sesame-foxes, 2022-11-11:
型号：x96max plus 2101w
芯片：jl2101
拆机图： 
[image]

现象：将盒子启动openwrt后插入路由器，会连接两次后速率降到100m，如果将盒子插入电脑则不会降速。但均不能正常工作。插路由器上无法看到盒子的ip，插电脑上也无法通过dhcp获取ip。
[image]

尝试过armbian，同样的现象。手动设置ip同样无法工作。

ophub, 2022-11-11:
https://github.com/unifreq/linux-5.15.y/blob/main/arch/arm64/boot/dts/amlogic/meson-sm1-x96-max-plus-q2.dts

q2和你的盒子没关系吧？

sesame-foxes, 2022-11-11:
> https://github.com/unifreq/linux-5.15.y/blob/main/arch/arm64/boot/dts/amlogic/meson-sm1-x96-max-plus-q2.dts
> 
> q2和你的盒子没关系吧？
> 
> 这是个百兆还是盒子？百兆试试meson-sm1-x96-air.dtb
> 
> 千兆的meson-sm1-x96-max-plus.dtb

meson-sm1-x96-max-plus.dtb无法启动，只有meson-sm1-x96-max-plus-q2.dtb可以正常启动。

sesame-foxes, 2022-11-11:
[image]
openwrt_official_s905x3_k5.10.154_2022.11.11.img
刚刚尝试了5.10也是一样

sesame-foxes, 2022-11-11:
> https://github.com/unifreq/linux-5.15.y/blob/main/arch/arm64/boot/dts/amlogic/meson-sm1-x96-max-plus-q2.dts
> 
> q2和你的盒子没关系吧？q2是换了qca9377蓝牙
> 
> 这是个百兆还是盒子？百兆试试meson-sm1-x96-air.dtb
> 
> 千兆的meson-sm1-x96-max-plus.dtb
dtb：meson-sm1-x96-max-plus.dtb
可以启动版本：openwrt_official_s905x3_k5.10.154_2022.11.11
无法启动版本：openwrt_lede_s905x3_k5.15.78_2022.11.11

经过多次尝试，5.10版本可以使用meson-sm1-x96-max-plus.dtb启动，5.15使用meson-sm1-x96-max-plus.dtb直接卡logo。
5.10使用meson-sm1-x96-max-plus.dtb启动后，直接插电脑，可以正常工作，电脑显示协商1000m（未实际测试）。如果插路由器，又会自动掉到100m。

ophub, 2022-11-11:
https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/armbian-docs/README.cn.md

最后12.12

这两个内核只差这个u-boot，你在5.15里复制一下，重启测试

ophub, 2022-11-11:
[image]

可以试试在dts里修改delay参数

sesame-foxes, 2022-11-11:
> https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/armbian-docs/README.cn.md
> 
> 最后12.12
> 
> 这两个内核只差这个u-boot，你在5.15里复制一下，重启测试

经过测试可以启动了，在5.15下依然存在5.10同样的问题，即连接电脑协商千兆，连接路由协商百兆。请问这是如何造成的？

[image]

sesame-foxes, 2022-11-11:
[image]
在电脑上可以正常千兆

ophub, 2022-11-11:
这个景略的驱动是f大春季的时候找厂家要的，不是内核自带的，用户较小没有更多地调试。你如果时间存足，到群里找他帮忙给你多测一下，找到合适的值。

sesame-foxes, 2022-11-11:
> 这个景略的驱动是f大春季的时候找厂家要的，不是内核自带的，用户较少没有更多地调试。你如果时间存足，到群里找他帮忙给你多测一下，找到合适的值。 你可以根据上面的标注自己调试一下。

可以尝试自行调试，这个数值有范围吗？要往大了调还是往小了调。

ophub, 2022-11-12:
0-3000，步进200

closed 2022-11-12 by ophub

sesame-foxes, 2022-11-12:
> 0-3000，步进200

用了一晚上的时间进行了尝试，0-3000全部尝试了一遍，依然无法和路由器协商到千兆。（两台路由器，小米r3p和WDR7650，均为千兆）
[image]
[image]
我的操作如图，是否存在操作错误？

ophub, 2022-11-12:
没错，改了dts要编译成dtb，替换你盒子里的/boot/dtb/amlogic目录下的同名文件

sesame-foxes, 2022-11-12:
> 没错，改了dts要编译成dtb，替换你盒子里的/boot/dtb/amlogic目录下的同名文件

已经替换了，暂时没有效果。我重新生成了30个，从100-3000,100为一个步进试一下。不行我就没法子了
[image]

ophub, 2022-11-12:
你白天在群里找他反馈一下景略JL2101驱动的测试情况，请他帮你多调试一下

sesame-foxes, 2022-11-12:
> 你白天在群里找他反馈一下景略JL2101驱动的测试情况，请他帮你多调试一下

好的，已经全部跑了一遍，全部都是协商百兆。我白天找他问一下