ophub/amlogic-s9xxx-openwrt #335: 软件包更新失败[openwrt_lede_s905l3a_k5.15.78_2022.11.12]

Edwardyx, 2022-11-16:
openwrt_lede_s905l3a_k5.15.78_2022.11.12.img.gz
<html>
<body>

固件版本 | OpenWrt R2022.11.11 / LuCI Master (git-22.314.43289-84299a5)

内核版本 | 5.15.78

</body>
</html>

[image]
软件包更新失败。直接访问上面的网址，可以浏览器可以正常打开下载。
我也不知道是不是设置了广告过滤的原因，就把广告屏蔽plus+和adguard都关闭了，更新还是失败。
一气之下firstboot，更新依旧失败。
宝盒里面升级到：openwrt_lede_s905l3a_k5.15.78_2022.11.13.img.gz还是失败
[image]

ophub, 2022-11-16:
错误信息显示，wget命令执行失败，提示查看网络设置。

Edwardyx, 2022-11-16:
我知道啊，但是，我不知道哪个地方设置有问题啊。可以正常上网，浏览器也能下载源里面的文件啊。为什么op就不行呢？
再说我都试过firstboot还原op，改了个固定ip就更新，也是失败。
刚刚升级今天的11.16版本，更新还是失败。
真不知道问题出在哪里啊。

ophub, 2022-11-16:
你在ttyd里用wget命令下载一个网络文件看看这个包能不能正常执行

closed 2022-11-20 by ophub