ophub/amlogic-s9xxx-openwrt #337: sunvell T95m

puccio823, 2022-11-16:
Hi opehub, I have a sunvell t95m tv box that I use with LibreElec, and it works, I was thinking of installing openwrt to try Docker and other programs, I think this box is not powerful enough for LibreElec,it only has a gigabyte of ram, and by now that version is starting to give problems
this after is the boot from usb of the file  **openwrt_official_s905_k5.10.154_2022.11.12.img** 

```
U-Boot` 2020.07-dirty (Jul 26 2020 13:40:04 +0200) hexdunp-gxbb

Model: hexdump usbkbd/hdni u-boot gxbb SoC: Amlogic Meson GXBB (S905) Revision if:e (13:1)
Net: 
Warning: ethernet@c9410000 (eth0) using randon MAC address - be:70:00:41:b7:4a
eth0: ethernet@c9410000 
starting USB...
Bus usb@c9000000: USB DUC2 
Bus usb@c9100000: USB DUCZ
scanning bus usb@c9000000 for devices... 1 USB Device(s) found 
scanning bus usb@c9100000 for devices... 1 USB Device(s) found 
            scanning usb for storage devices... 0 Storage Device(s)
##Error inserting "stdin" variable, errno 22 
Card did not respond to voltage select!
Card did not respond to voltage select! 
Card did not respond to voltage select!
Card did not respond to voltage select!
Hit any key to stop autoboot: 0
Card did not respond to voltage select!
Card did not respond to voltage select!

Device 0: unknown device 
Speed: 100, full duplex
BOOTP broadcast 1
DHCP client bound to address 192.168.178.71 (109 ms)
www Warning: no boot file name: using 'COA8B247.ing" 
Using ethernet@c9410000 device
TFTP from server 192.168.178.1: our IP address is 192.168.178.71
Filename 'C0A8B247.ing'.
Load address: 0x1000000 
Loading: TTTTTTTTTT
Retry count exceeded: starting again 
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/COA8B247
Speed: 100, full duplex 
Using ethernet@c9410000 device
TFTP from server 192.168.178.1: our IP address is 192.168.178.71
Filename 'pxelinux.cfg/Con88247'. 
Load address: 0x1000000
Loading: TTTTTTTTTT
Retry count exceeded: starting again 
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/COABB24 
Speed: 100 full duplex 
Using ethernet@c9110000 device
TFTP from server 192.168.178.1: our IP address is 192.168.178.71 
Filename 'pxelinux.cfg/COA8B24'.
Load address: 0x1000000
Loading: TTTTTTTTTT
```

puccio823, 2022-11-16:
thanks for the quick reply
in this site they say that the cpu is S905
[https://www.savagemessiahzine.com/sec__/forum/index.php?showtopic=730321](url)
this is what i see
```

processor       : 2
processor       : 3
Features        : fp asimd evtstrm crc32 wp half thumb fastmult vfp edsp neon vfpv3 tlsi vfpv4 idiva idivt
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : Amlogic
Serial          : 00000000000000000000000000000000
```
for the network card I could only find this information

```
T95LibreELEC:~ # ethtool -p eth0
Cannot identify NIC: Operation not supporterd
T95LibreELEC:~ # ethtool -i eth0
driver: st_mac100
version: March_2013
firmware-version:
expansion-rom-version:
bus-info:
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
suorts-priv-flags: no
[1]+  Done(127)                  lspci
```
for the dtb file, I think I used the file at this link, I did it a long time ago, maybe now I wouldn't be able to anymore
[https://forum.libreelec.tv/thread/1223-s905-device-trees/?pageNo=3](url) scroll down the page, up to 8zabi

ophub, 2022-11-16:
https://m.gearbest.com/tv-box-mini-pc/pp_366029.html

This seller's introduction is s905x

Your own box, with the exact logo on the motherboard's soc.

Your NIC is on the front of the motherboard.

Don't guess what dtb you use, look at the configuration file in the LibreElec system.

The information you see in unsupported buggy firmware is only misleading.

Generally speaking, any device that can use LibreElec can try armbian

If you don't have accurate information, you can't guess which firmware to use

puccio823, 2022-11-17:
Thanks for your interest
I disassembled the box, but with my phone the resolution is not optimal, also on the soc there is an aluminum heatsink that does not allow reading, do you know how to remove it?
meanwhile, I am attaching this photo
[image]
[image]

ophub, 2022-11-17:
Is the SOC and other models marked on the bottom of the box?

puccio823, 2022-11-17:
I didn't quite understand which box you are talking about
the packaging box or some label on the bottom of the box?
I have done many tests with many dtb
and before that with this image **openwrt_official_s905x_k5.10.154_2022.11.16.img.gz** 
with the image for soc s905x the screen remains black

ophub, 2022-11-17:
Which dtb file are you using in LibreElec?

puccio823, 2022-11-17:
if it can be useful, i remember the version i downloaded for libreelec was for s905 i still have rar files
**LibreELEC-S905.arm-8.2-8.2.3.1-temp_sensor_disabled.img.gz**
to update kodi from android i used this ipk
**kodi-17.5.1-Krypton-armeabi-v7a.apk**
the dtb file I used that the one I downloaded from this link
[https://forum.libreelec.tv/thread/1223-s905-device-trees/?pageNo=3](url)

ophub, 2022-11-17:
https://github.com/ophub/amlogic-s9xxx-armbian/tree/main/build-armbian/common-files/bootfs/dtb/amlogic

These two systems use similar dtb names. Don’t you know what dtb you used in the LibreELEC system before?

puccio823, 2022-11-17:
i found this dtb file in the documents folder **gxl_p212_1g.dtb**
yes I have already done a lot of tests with various dtb files
also the one downloaded from the libreelec site
that's why I wrote here
ah I forgot libreelec starts from microSD
the tests with openwrt are done with usb
and it often happens that every time I do a test I am then forced to fix the box because it does not start, luckily I have a recovery image

ophub, 2022-11-17:
meson-gxl-s905x-p212.dtb

puccio823, 2022-11-17:
ok, now that i think about it since i used the image for s905 i never did any tests with dtb for s905x, i will try after lunch
now I'm going to pick up the kids from school
I'll let you know later

puccio823, 2022-11-18:
hi last night i did some tests and finally it booted with **meson-gxbb-p200.dtb** file
[image]
but now I have this problem
[image]

ophub, 2022-11-18:
Ok, congratulations you found the right configuration.
According to previous user feedback, the s905 box does not support writing to emmc, and can only be used on USB/TF/SD cards.

puccio823, 2022-11-18:
maybe you didn't notice that i have no ethernet, i have no connection and no WiFi either

ophub, 2022-11-19:
https://github.com/LibreELEC/linux_openvfd/blob/master/conf/meson-gxl-s905x-t95m.conf

From this led configuration file, you are s905x

puccio823, 2022-11-19:
i didn't quite understand what to do now, that file is used for libreelec, do i have to do the same for openwrt? if the box does not have access to the network, it is not useful

ophub, 2022-11-19:
You can't be sure what your soc is, only test these dtbs.

ophub, 2022-11-20:
The armbian system in the USB can only be started on the basis of the Android system.

If the libreelec system is installed in the current box, first backup your libreelec system, and then restore to the original Android system. so that you can test armbian

puccio823, 2022-11-20:
hi openhab, yesterday i tried to fix the front panel, to no avail, the box starts with the **meson-gxbb-p201.dtb** file
and with this the ethernet cable also works, but not the wifi
no card is detected
the leds must also be started from http://192.168.178.101/cgi-bin/luci//admin/system/leds?
for the front panel i tried some files
here https://github.com/LibreELEC/linux_openvfd/blob/master/conf/meson-gxl-s905x-t95m.conf
I uploaded them and renamed them to diy.conf
in some I have also made changes as described in this guide
https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/armbian-docs/led_screen_display_control.md
nothing I could not turn on the panel
where am i wrong?

ophub, 2022-11-20:
meson-gxbb-p201.dtb

Wired internet works just fine. Because it does not add wireless and led related codes in its dtb, so other tests are invalid.
What is your wireless chip?

puccio823, 2022-11-20:
I have no idea what it could be, if I start libreelec can I know ?, give me time

ophub, 2022-11-20:
The wireless chip is on your motherboard, look carefully at the code on the chip

puccio823, 2022-11-20:
here I am, sorry for the delay but we have a different time zone :)
looks like realtek 8189ETV

[image]

ophub, 2022-11-21:
https://github.com/unifreq/linux-5.15.y/tree/main/drivers/net/wireless/realtek

There is no driver for 8189ETV in the kernel.

ophub, 2022-11-21:
Do you use the 5.10 kernel? Take a look at which u-boot you are using. Use the following command to send a screenshot
```
cat /boot/uEnv.txt
cd /boot
sha256sum u-boot*
```

puccio823, 2022-11-21:
good morning ophub
here is the screen
[image]

ophub, 2022-11-21:
Do you use the 5.10 kernel? Use the following command to send a screenshot.

```
cd /boot
sha256sum u-boot*
```

puccio823, 2022-11-21:
ok
`login as: root
root@192.168.178.100's password:
 
 Install OpenWrt: System → Amlogic Service → Install OpenWrt
 Update  OpenWrt: System → Amlogic Service → Online  Update
 Amlogic Box SoC: s905x | OpenWrt Kernel: 5.15.78-flippy-78+o
 Production Date: 2022-11-16
───────────────────────────────────────────────────────────────────────
 Device Model: Amlogic Meson GXBB P201 Development Board
 Architecture: AArch64 : Cortex-A53 x 4
 Load Average: 0.38 0.37 0.20   Uptime: 14m 23s
 Ambient Temp: 4294967295.0  °C         CPU Freq: 2016 Mhz
 Memory Usage: 20% of 803M      Swap Usage: 0% of 0M
 Boot Storage: 40% of 251.0M    ROOTFS: 87% of 960.0M
 Data Storage: 3% of 5.3G       IP Addr: 192.168.178.100
───────────────────────────────────────────────────────────────────────
root@OpenWrt:~# cd /boot
root@OpenWrt:/boot# sha256sum u-boot*
fb0d8e321828642bf2c930dae96fd0048933f87e4d4228a77270a4d1aa7e7b41  u-boot-e900v22                                                                                        c.bin
7c0c91e60d107c61de03798fcb04e462c7e5616b400ee6abc096b94e97a2cee0  u-boot-gtking.                                                                                        bin
b8fc82e1a4a72ce15ee6fee8776ef26f9ae4834b641a2e4b6a912a3e89efdff6  u-boot-gtkingp                                                                                        ro-rev-a.bin
215f2f3abbd03f19a7b304f5dd7824914a16d9bdecb0711845a8c2fa7e292483  u-boot-gtkingp                                                                                        ro.bin
5094d8144688c5fc20424497afb11b04187c9efdf8adf95cd2a9fbbd951b8cfc  u-boot-n1.bin
e0e7a258e024aa8e825a6c46c68fdf0e6da2dfa54d379115b91fae25176e748f  u-boot-odroid-                                                                                        n2.bin
3a76dc9b2d80988ffbaade9555700b6d9121498eef6f75000ebd11e9f991ed3f  u-boot-p201.bi                                                                                        n
c3b2065356e61cec05320e68010135a315e7d89d0e6d6dd212a55a28cf90f7e8  u-boot-p212.bi                                                                                        n
f69f6241224f72e6942119dae0d026154089cdff8442d5a4c93de4d8bc3e69b9  u-boot-s905.bi                                                                                        n
3becd7d97afaa7fbcb683eb4c28221f282bf73b74d71138ac6be768611f8e11f  u-boot-s905x-s                                                                                        912.bin
c3a3662453cbbfcd7a11e2c829017b667e601708f2f5c85543f1727249787a74  u-boot-s905x2-                                                                                        s922.bin
5ff0be52537bd01ab6aa772e0ee284ab1e1f47f43cbc08da2a6c9982ef1df379  u-boot-sei510.                                                                                        bin
ddf6ff930c13c03528b64a738de1126f811f6802c5096495ef83f6c07c4986d1  u-boot-sei610.                                                                                        bin
b8e649dc39e2e869a32bcebde06d6e604e5107e03a9d60f04c453081ebf0885c  u-boot-skywort                                                                                        h-lb2004.bin
3e89947c31bacc31574213f86ab03339291dcde928872dc6ae957fce58e76ac9  u-boot-tx3-bz.                                                                                        bin
aba80167498ef01482118c210636f14306ccaea80d09dce9e4358b993c0b3d88  u-boot-tx3-qz.                                                                                        bin
065370807a82019677cff7adaea917f1a08c5cc4baca4c5d9e19b117b66e5ce6  u-boot-u200.bi                                                                                        n
74eb7d04eb743b5e512dcf3af42061a5b90306afcd46f64bdea427e18812b2a1  u-boot-ugoos-x                                                                                        3.bin
e23bc57cefb1b99ecfb7a4192e8960c9b2a44f75d902ad6ff007108f9e01cc7b  u-boot-x96max.                                                                                        bin
2bc15470d83f9e4e748f62897a0b3f71f896da51bf41591a03eebc289ee703dc  u-boot-x96maxp                                                                                        lus.bin
3df7343e56116244b2d2d2fa8bcdbf411c088667bfc850f163d3b0b8caca29aa  u-boot-zyxq.bi                                                                                        n
f93143df439579216bdf7f6264b2b5169d177cc3cfd0aa43503feac9ff298242  u-boot.sd
edadc4e10b63eddbb82166b756f915ff27cfd2ade9012d0afd0bdbcae09cc8f0  u-boot.usb
root@OpenWrt:/boot#
`

ophub, 2022-11-21:
Temperature error. Use the following command to send a screenshot.

```
sensors | grep -Eo '\+[0-9]+.+C' | sed ':a;N;$!ba;s/\n/ /g;s/+//g'

awk '{ printf("%.1f °C", $0 / 1000) }' /sys/class/hwmon/hwmon0/temp1_input

awk '{ printf("%.1f °C", $0 / 1000) }' /sys/class/thermal/thermal_zone0/temp
```

puccio823, 2022-11-21:
i am using kernel 5.15
[image]

ophub, 2022-11-21:
At this temperature, steel can be made and electricity can also be generated. I think the display needs to be corrected. I'll update the script in a while.

puccio823, 2022-11-21:
> A questa temperatura si può produrre acciaio e generare elettricità. Penso che il display debba essere corretto. Aggiornerò lo script tra un po'.

I think the problem is where the "."
actually it would be 42.94

ophub, 2022-11-21:
```
# Download the latest script
wget https://raw.githubusercontent.com/ophub/amlogic-s9xxx-openwrt/main/amlogic-s9xxx/common-files/patches/cpustat/30-sysinfo.sh -O etc/profile.d/30-sysinfo.sh
```

You download the latest script in ttyd, then log in again to see the temperature display, and post the screenshot

puccio823, 2022-11-21:
ok 
[image]

[image]

ophub, 2022-11-21:
Okay, perfect, the debugging of this box is over for the time being, please feel free to report back if you have any questions in the future.

puccio823, 2022-11-21:
I would need some tutorials on how to install the containers docker, I tried but I can install the containers, it starts but I don't see the web page

ophub, 2022-11-21:
run: `armbian-software`

closed 2022-11-22 by ophub