ophub/amlogic-s9xxx-openwrt #338: s905 sunvell q-box emmc

suezz, 2022-11-18:
Hi!, sorry my noob question..., does it possible the installation on emmc with the dtb I'm using in libreelec?
gxbb_p200_2G_1Gbit.dtb

tried using s905_k5.15.78_2022.11.16.img with  meson-gxbb-beelink-mini-mx.dtb  but no success
or no way for s905?

I ask before have to restore from brick another time....:)

ophub, 2022-11-19:
meson-gxbb-p200.dtb
meson-gxbb-p201.dtb

suezz, 2022-11-20:
thanks for reply but
brick
brick

is there a way to build the dbt from the one that is ok for libreelec?
or other to try? I make a new usb cable and became fast in restore android...:))

ophub, 2022-11-20:
The armbian system in the USB can only be started on the basis of the Android system.

If the libreelec system is installed in the current box, first backup your libreelec system, and then restore to the original Android system. so that you can test armbian

suezz, 2022-11-21:
Specific that I'm trying with OpenWRT, I see that booting from USB works whether Android or LibreELEC is installed in the box. I can't do the installation on emmc... I realized that it is quite complicated and it is already a lot that works from USB without modifications. The 'problem' is that if you have also connected a tethering smartphone the box does not boot from USB... Thanks all the same! :)

ophub, 2022-11-22:
Currently s905 TV Boxes can only be used in TF/SD/USB, other types of TV Boxes also support writing to EMMC.

closed 2022-11-22 by ophub