ophub/amlogic-s9xxx-openwrt #339: Ugoos AM6B Plus (2021)

cleberjorge, 2022-11-21:
TV Box Ugoos AM6B Plus (2021) WiFi is not recognized using openwrt_official_s922x_k5.15.79_2022.11.20.img image and DTB meson-g12b-ugoos-am6b-plus.dtb

ophub, 2022-11-21:
no wifi6 driver

cleberjorge, 2022-11-21:
I use EmuELEC in the Box and the Wifi works perfectly, I tried to use the EmuELEC DTB and Openwrt but it didn't work.
Perhaps this information will be useful to you.
Thanks for the feedback

cleberjorge, 2022-11-21:
> no wifi6 driver

I use EmuELEC in the Box and the Wifi works perfectly, I tried to use the EmuELEC DTB in Openwrt but it didn't work.
Perhaps this information will be useful to you.
Thanks for the feedback

ophub, 2022-11-21:
The Android system has drivers, but there are no drivers in the mainline kernel.

closed 2022-11-22 by ophub