ophub/amlogic-s9xxx-openwrt #341: 大神好，我又来了

281677160, 2022-11-22:
[image]

[image]

````
      - name: Initialization environment
        id: init
        env:
          DEBIAN_FRONTEND: noninteractive
        run: |
          docker rmi `docker images -q`
          sudo -E apt-get remove -y --purge azure-cli ghc* zulu* llvm* firefox google* powershell openjdk* msodbcsql17 mongodb* moby* snapd* mysql*
          sudo rm -rf /etc/apt/sources.list.d/* /usr/share/dotnet /usr/local/lib/android /usr/lib/jvm /opt/ghc /swapfile
          sudo -E apt-get -qq update -y
          sudo -E apt-get -qq full-upgrade -y
          sudo -E apt-get -qq install -y $(curl -fsSL https://is.gd/depend_ubuntu2204_openwrt)
          sudo -E apt-get -qq autoremove --purge
          sudo -E apt-get -qq clean
          sudo timedatectl set-timezone "${TZ}"
          sudo mkdir -p /workdir
          sudo chown ${USER}:${GROUPS} /workdir
          echo "status=success" >> ${GITHUB_OUTPUT}
````
大神你自己测试看看吧，希望对您有用

ophub, 2022-11-22:
https://github.com/coolsnowwolf/lede/blob/master/.github/workflows/openwrt-ci.yml

我是前些天刚更新的，从大雕家直接复制粘贴的，你对比下。
-y 这个参数在本地要加，但是github上可以不加，全部他都会自动回车执行。

281677160, 2022-11-22:
哦，我也没看他的，我就今天看到您的，然后测试了一下，那个php不能删除，删除了会一直报错

11月22日12:16:04 fv-az413-557 php-fpm8.1[80306]：[22-Nov-2022 04:16:04]错误：未能加载配置文件“/etc/php/8.1/fpm/php-fpm.conf”

11月22日12:16:04 fv-az413-557 php-fpm8.1[80306]：[22-Nov-2022 04:16:04]错误：FPM初始化失败

11月22日12:16:04 fv-az413-557 systemd[1]：php8.1-fpm。服务：主进程已退出，代码=已退出，状态=78/CONFIG

这是我百度翻译翻译过来的，一直显示

还有那个删除hhvm跟adoptopenjdk*这个，直接就报错停止了，除非|| true，呃，我也不知道他干嘛要这样搞，没弄懂

 
https://github.com/coolsnowwolf/lede/blob/master/.github/workflows/openwrt-ci.yml

ophub, 2022-11-22:
编译openwrt不需要这些，所以Actions的服务器可以删除。很多人需要编译高大全版，80G的空间不够用，所以能删尽删，腾出更大的空间用于编译。

你本地编译话，如果你需要使用php，就不要把自己电脑的php删除了，按照本地依赖安装说明，只安装不删除即可。
https://github.com/ophub/amlogic-s9xxx-openwrt#local-build-instructions

281677160, 2022-11-22:
这个我本地没删除这些，我是云编译测试的，就是云编译删除php不行的，你不相信，你不要用用隐藏日志方式运行就明白了
就是全部 sudo -E apt-get -qq  改成 sudo apt-get 你会看到一个一直显示的错误，说实在我也不知道有没有影响，所以才来跟您说一声的
我以前一直用天灵那个集成好的一键链接安装依赖的，昨天本地打包发现不会压缩，因为您的代码错误了呢，哈哈
说实在，天灵的这个很好用  sudo bash -c 'bash <(curl -s https://build-scripts.immortalwrt.eu.org/init_build_environment.sh)'

 

 
 https://github.com/ophub/amlogic-s9xxx-openwrt#local-build-instructions

ophub, 2022-11-22:
https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/3514812579/jobs/5889290689
你可以看下我已有的Actions运行结果，都顺利执行完成了，PHP在github默认的系统都没带

ophub, 2022-11-22:
https://www.cnblogs.com/zouhong/p/16672312.html

pigz打包解包的速度比gzip快一倍。
因为要打包34个文件，使用更快的方法能节约时间。
很多编译高大全的用户，都踩着6小时的红线跑，时而就超时了。
所以节约时间，节约服务器空间，是使用github Actions需要考虑的。

281677160, 2022-11-22:
哈哈，感谢大神弄了这么好的打包程序，方便快捷

 
https://www.cnblogs.com/zouhong/p/16672312.html

ophub, 2022-11-22:
https://github.com/breakings/OpenWrt/actions

你看看这个高手的打包，好几个失败的原因是6小时到了。
前几天失败是因为空间满了。

因为我的这个打包选的插件比较少，所以感觉不到超时和空间不足的问题，但是一般用户都需要在这个fork上加很多软件包。所以本着可以用不到，不能不精算的原则，我也是照着他们的方法尽量追求机制。

对了，上面这个老哥的仓库会更适合你使用，这是高大全的配置版本，维护很及时。

ophub, 2022-11-22:
https://github.com/ophub/amlogic-s9xxx-openwrt/commit/7a8961a72f56c17a8e270b77025dcd92555aa109

php、hhvm跟adoptopenjdk*都取消删除了。

closed 2022-11-22 by ophub

ophub, 2022-11-22:
https://github.com/ophub/amlogic-s9xxx-openwrt/commit/c4a78fc8d91196c1fc9fbe109c2c08d9a53ffd5f

已经按你的测试更新了