ophub/amlogic-s9xxx-openwrt #344: x96-air no terminal or no display, tried x96 air gbit and simple dtbs

P9ison, 2022-12-03:
- SOC: S905x3
 - Model: X96-Air

Tried booting with both dtbs, one of gbit and one simple. I am not getting any visual display at all, but can confirm that it does boot because i tried plugging in the ethernet cable and my adapter showed me gateway of 192.168.1.1 but i cannot login or ssh into 192.168.1.1 (No route to host).
a

ophub, 2022-12-04:
If you already have a 192.168.1.1 device in your network, connect the x96 to the computer, otherwise the ip will re-emerge.

openwrt does not require hdmi

P9ison, 2022-12-04:
#

> If you already have a 192.168.1.1 device in your network, connect the x96 to the computer, otherwise the ip will re-emerge.
> 
> openwrt does not require hdmi

It doesn't work i cannot access luci or ssh just can see the gateway. And no display.

ophub, 2022-12-05:
/boot/uEnv.txt

X96-Air-1Gb: meson-sm1-x96-air-gbit.dtb
X96-Air-100Mb: meson-sm1-x96-air.dtb
X96-Air-Q1000: meson-sm1-x96-max-plus-q2.dtb

P9ison, 2022-12-05:
> /boot/uEnv.txt
> 
> X96-Air-1Gb: meson-sm1-x96-air-gbit.dtb X96-Air-100Mb: meson-sm1-x96-air.dtb X96-Air-Q1000: meson-sm1-x96-max-plus-q2.dtb

Do i have to change the Uboot for any of these files?
And should I try getting a display output with AV Cable?
One thing i forgot to mention was, I was not getting any display at all after the x96 air splash logo it went no signal. So no black screen or anything.

ophub, 2022-12-05:
try 5.15.y.
don’t need u-boot, only dtb right.

P9ison, 2022-12-05:
> try 5.15.y. don’t need u-boot, only dtb right.

Ok i tried it now i get stuck at device eth0 entered promiscuous mode.

P9ison, 2022-12-05:
[image]

P9ison, 2022-12-05:
[image]

Ok now it says this for some reason

ophub, 2022-12-06:
This is the normal startup, press Enter on the keyboard, and you can see the welcome interface. Now the browser can access 192.168.1.1 to manage

P9ison, 2022-12-06:
> This is the normal startup, press Enter on the keyboard, and you can see the welcome interface. Now the browser can access 192.168.1.1 to manage

Alright thanks.
How do I reset to defaults?

P9ison, 2022-12-06:
> ttyd or ssh: `firstboot`

Hey i cannot upload anything.
Like i have tried SCP, FTP but it stays at 0% and creates a 0kb file of whatever i am trying to upload.

closed 2022-12-06 by P9ison

ophub, 2022-12-07:
can use scp or web upload