ophub/amlogic-s9xxx-openwrt #345: hk1 fail to boot after flash to EMMC

BobDLA, 2022-12-04:
Standard chat template, no routine, no chat.
标准聊天模板，无套路不聊天。

**Device Information | 设备信息**
 - SOC:  S095x3
 - Model HK1

**Armbian Version :openwrt
 - Kernel Version: e.g. k5.15.80
 - Release:   openwrt_lede_s905x3_k5.15.80_2022.12.03.img.gz
                    openwrt_official_s905x3_k5.15.80_2022.12.01.img.gz

**Describe the bug | 问题描述**
Describe the problem in detail and attach screenshots if necessary.
详细描述问题，并在必要时附上屏幕截图。

The HK1 could boot up after I flash the openwrt in the USB stick.
But , after I flash the system to EMMC follow the instruction. The HK1 failed to boot up.
No HDMI output and the computer conned to HK1 failed to get IP address. 
I test both lede and official version, same issue. 

Any suggestion?

Thanks

ophub, 2022-12-04:
You try the 5.10 kernel firmware, he has one more u-boot file than 5.15.

You can also add tests manually in 5.15
```
cd /boot
# use in usb
cp -f u-boot-x96maxplus.bin u-boot.ext
# use in emmc
cp -f u-boot-x96maxplus.bin u-boot.emmc
```

BobDLA, 2022-12-04:
> You try the 5.10 kernel firmware, he has one more u-boot file than 5.15.
> 
> You can also add tests manually in 5.15
> 
> ```
> cd /boot
> # use in usb
> cp -f u-boot-x96maxplus.bin u-boot.ext
> # use in emmc
> cp -f u-boot-x96maxplus.bin u-boot.emmc
> ```

Thanks for you info. I have tried the version, but it's still not working.
And then, I reflash the EMMC to a factory Android system, then,  install openwrt by USB again.
It's working now.

I have installed armbian in the EMMC before, and that may cause the issue?

And, thanks for your great project and help gain.

ophub, 2022-12-04:
The boot is adapted to the original Android system, and it will be affected if other systems have been installed before.
You solved it very well.

closed 2022-12-04 by ophub