ophub/amlogic-s9xxx-openwrt #347: Cannot enable wifi on X96 air kernel 5.15.80 & 81.

P9ison, 2022-12-06:
- SOC: S905X3
 - Model X96 Air 
  
Wifi does not enable on kernel 5.15.81 and 80
It works perfectly on 5.10.156
When i tap enable on 15.81 & 15.80 
Box reboots.

ophub, 2022-12-07:
This driver comes from a third party, and it is not very compatible with the mainline kernel. Some versions will not work. You should use 5.10.

closed 2022-12-07 by ophub

hillz2, 2023-01-03:
I have X96 Air too and wifi doesn't work, where can I find the download link for kernel 5.10.156 @P9ison @ophub ?

ophub, 2023-01-03:
armbian-update -k 5.10.161

hillz2, 2023-01-03:
@ophub my OpenWrt doesn't have armbian-update installed

ophub, 2023-01-03:
Login in to openwrt → system menu → Amlogic Service → Select the 5.10 kernel in the settings  →  Download updates online