ophub/amlogic-s9xxx-openwrt #348: Abusing git rebase

darkliang, 2022-12-07:
Hi, thank you for your hard work. However, I think using rebase without merge may cause some problems, such as changing the order of commits, losing previous commit information, and potential conflicts if working with others. I suggest using merge during development to ensure the integrity and consistency of the project's commit history.

ophub, 2022-12-07:
Because many binary files and large compressed packages are saved The git file will continue to grow. According to github, it will be difficult to maintain when it is larger than 2G. Therefore, I am used to using `git filter-repo` to regularly clean up obsolete files and directories, which will lose part of the commit, but it is necessary for limited space.

darkliang, 2022-12-07:
Maybe you can move the binary files to an independent repo. In this repo, only the scripts are maintained.

ophub, 2022-12-07:
Now it is being optimized and has been classified into different repo for management.

darkliang, 2022-12-07:
That's great. I propose this issue because I find that the repo I forked several days ago has hundreds of deviated commits with this upstream repo. This makes me hard to sync the latest code.

ophub, 2022-12-07:
I also found this problem. If you delete a history file (which is obsolete and can be completely cleaned up), the tool deletes a commit, and subsequent commits will deviate.

darkliang, 2022-12-07:
Anyway, it's not good to destroy the consistency of the commit history

ophub, 2022-12-07:
This tool is recommended by github. I learned from their recommendations. If you know there are good management tools, you can recommend them.

closed 2022-12-07 by ophub