ophub/amlogic-s9xxx-openwrt #349: 你好作者

LEGION128, 2022-12-08:
我是魔百盒cm201-1ys 广东移动的盒子  请问这款盒子可以刷openwrt吗？  如果可以我需要下那个固件 
[image]

ophub, 2022-12-09:
/boot目录的uEnv.txt改dtb名称为meson-gxl-s905x-p212.dtb
把u-boot-s905x-s912.bin复制为u-boot.ext

LEGION128, 2022-12-10:
你好 是要下载这个固件吗？
[image]

ophub, 2022-12-10:
对

LEGION128, 2022-12-10:
[image]

ophub, 2022-12-10:
复制一份，原来的保留

LEGION128, 2022-12-10:
不是很懂

LEGION128, 2022-12-10:
好像懂了 吧固件刷到u盘 虚插到盒子的usb孔上 通电开机马上吧u盘插进 屏幕显示花瓶就是代表成功了是吧

ophub, 2022-12-10:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router-config/README.cn.md

10.10

LEGION128, 2022-12-10:
您好 根据你的提示 我在盒子上开启了usb调试 win上下载且配置了adb  但是我填入ip显示这个
[image]

ophub, 2022-12-10:
你这没adb连接进去，继续尝试

closed 2022-12-13 by ophub

Cxy-2008, 2023-05-13:
你好，我是跟你同款盒子，也是套装124的，请问你解决了吗？