ophub/amlogic-s9xxx-openwrt #351: Tanix Tx5 Max (S905x2)

mreser99, 2022-12-09:
I wanna use openwrt official in this device. Is that possible?

Specs:
SoC: Amlogic S905X2 Quad Core
RAM and Storage: 4 GB DDR4 / 32 GB eMMC
[image]

ophub, 2022-12-09:
try s905x2-km3

mreser99, 2022-12-09:
Thank you it's booted but i couldn't connect web-ui at all. Am i doing something wrong?
When i check ip address it seems exist.

ophub, 2022-12-09:
Do you have a monitor and keyboard? Connected, after the startup code stops, press Enter to see the welcome interface, right?

ophub, 2022-12-09:
cat /boot/uEnv.txt

mreser99, 2022-12-09:
No i'm using `openwrt_official_s905x2_k5.10.158_2022.12.09.img` `km3` one had ethernet issue. 
This is the screenshot and yes i'm using with default options.

[image]

ophub, 2022-12-09:
Install option 301 - X96-Max-4GB

mreser99, 2022-12-09:
These are for just install or these gonna make my wifi works also?
It doesn't make my device brick don't it? 
I'll do this like this, right?
[image]

And thank for your help 😊

ophub, 2022-12-09:
right

mreser99, 2022-12-09:
Sorry dear developer it's my fault. 
Here it is;
`Network: Wifi ac 2×2 (RTL8222BS) – Bluetooth 4.1 – Ethernet Gigabit`

ophub, 2022-12-09:
https://github.com/unifreq/linux-5.15.y/tree/main/drivers/net/wireless/realtek

no drivers.

mreser99, 2022-12-09:
I see dear developer. I'll buy something with compatible. Thank you for all :)

closed 2022-12-13 by ophub