ophub/amlogic-s9xxx-openwrt #354: LEDE in English

mreser99, 2022-12-13:
Is there any way to use LEDE project with English Language?

ophub, 2022-12-14:
Add English translation files to the LEDE-plugin

mreser99, 2022-12-14:
How can i do that, into .img or somewhere else? And how can i find the translation files as you mentioned?

ophub, 2022-12-14:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config

ophub, 2022-12-14:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config#411-let-the-firmware-support-the-native-language

If the plug-in you use has an English translation, just enable it. Generally, plug-ins come with English language packs.

closed 2022-12-14 by ophub

mreser99, 2022-12-14:
Plug-ins are yes but whole menu of LEDE is chinese. I'll try

mreser99, 2022-12-15:
I've made 2 different builds since yesterday but no luck, still chinese. I've removed chinese and put the english main language but it's not resolved. 
Did i something wrong?

ophub, 2022-12-16:
screenshot