ophub/amlogic-s9xxx-openwrt #355: 编译固件IP地址默认是DHCP修改哪里吗？

3136083906, 2022-12-18:
Standard chat template, no routine, no chat.
标准聊天模板，无套路不聊天。

**Device Information | 设备信息**
 - SOC: [e.g. S912]
 - Model [e.g.章鱼星球]

**Armbian Version | 系统版本**
 - Kernel Version: [e.g. 5.15.100]
 - Release: [e.g. lede]

**Describe the bug | 问题描述**
Describe the problem in detail and attach screenshots if necessary.
详细描述问题，并在必要时附上屏幕截图。

固件的默认地址是DHCP可以吗？编译固件时怎么修改哪里？

ophub, 2022-12-19:
https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/armbian-files/common-files/etc/network/interfaces

现在默认的就是dhcp分配ip

3136083906, 2022-12-19:
> https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/armbian-files/common-files/etc/network/interfaces
> 
> 现在默认的就是dhcp分配ip

我的意思是做个旁路由，主路由器分配IP，这个固件没有dhcp

ophub, 2022-12-19:
我看串了，我以为你问armbian

/etc/config/dhcp

自己添加这个文件固定

3136083906, 2022-12-19:
> 我看串了，我以为你问armbian
> 
> openwrt是 /etc/config/dhcp
> 
> 自己添加这个文件固定
> 
> https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/common-files/rootfs
> 
> 这里就是根目录，想固定的文件按照openwrt的目录及文件名放入，打包时将合并这些固定的个性化文件

好的，请问不想要固件集成docker修改哪里

ophub, 2022-12-19:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config/lede-master

配置里去掉

3136083906, 2022-12-19:
> https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config/lede-master
> 
> 配置里去掉

OK

closed 2022-12-19 by ophub