ophub/amlogic-s9xxx-openwrt #356: 更新错误

sollyu, 2022-12-22:
Standard chat template, no routine, no chat.
标准聊天模板，无套路不聊天。

**Device Information | 设备信息**
 - SOC: S905L3
 - Model: 魔百盒

**Armbian Version | 系统版本**
 - Kernel Version: 5.15.76
 - Release: lede

**Describe the bug | 问题描述**
官方没有`S905L3`的包，我用的是`S905L3A`的包，使用`meson-gxl-s905l2-x7-5g.dtb`就可以正常启动。
但是我尝试更新系统提示下面的错误
[image]
我看了一下这个文件是存在的，并且文件的内容如下：
```bash
PLATFORM='amlogic'
KERNEL_VERSION='5.15.76'
K510='0'
FDTFILE='meson-gxl-s905l2-x7-5g.dtb'
UBOOT_OVERLOAD=''
MAINLINE_UBOOT=''
ANDROID_UBOOT=''
SOC=''
```
请问我该如何操作才能升级？

ophub, 2022-12-22:
你的固件有问题
UBOOT_OVERLOAD=''
SOC=''

不对，这两个必须有值，你从哪下载的固件？

sollyu, 2022-12-22:
从本项目下载的，就只处理了`uEnv.txt`其他未动，然后输入到emmc里的。

ophub, 2022-12-22:
我这下的固件以下这两个不可能是空，必定有/lib/u-boot这个字，你是fork改的吧，把model改错了。
MAINLINE_UBOOT=''
ANDROID_UBOOT=''

sollyu, 2022-12-22:
没啊，天地良心，真的没改。那我现在这种情况还能抢救一下吗？

ophub, 2022-12-22:
给SOC添加值：s905l3

sollyu, 2022-12-22:
我的盒子的型号`魔百盒 CM311-1 YS(M)`拆开之后看芯片是`S905L3`后面没有`A`，最新的`meson-gxl-s905l2-x7-5g.dtb`是可以正常启动的（大约半年前开始支持的）

现在好像可以能更新了，但是GitHub的网不是很稳定，有手动下载文件的办法吗？

[image]

下面是我这个盒子背面的照片

[image]

sollyu, 2022-12-22:
我手动上传更新文件，然后点击更新，出现了这个错误：

[image]

ophub, 2022-12-22:
`lsblk`
看下你的磁盘情况
`ls /mnt/*4/`

sollyu, 2022-12-22:
刚才确实是挂在的盘有问题，但是新的问题又出现了

这个是现在的磁盘情况

```text
root@OpenWrt-S905L3 /m/mmcblk2p4# lsblk

NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
mmcblk2      179:0    0 14.6G  0 disk
├─mmcblk2p1  179:1    0  160M  0 part
├─mmcblk2p2  179:2    0  960M  0 part /
├─mmcblk2p3  179:3    0  960M  0 part
└─mmcblk2p4  179:4    0 12.3G  0 part /mnt/mmcblk2p4
mmcblk2boot0 179:32   0    4M  1 disk
mmcblk2boot1 179:64   0    4M  1 disk
```

[image]

ophub, 2022-12-22:
你删除什么插件时，把lsblk命令也删了，缺少这个必要依赖无法继续。你在软件中心，先更新，在搜索lsblk安装

ophub, 2022-12-22:
你选择的配置是113的，你下次安装时，选择113安装即可，别手动输入了。

sollyu, 2022-12-22:
刚才下载了最新的镜像文件重新刷系统，下载了[openwrt_lede_s905l3a_k5.15.84_2022.12.21.img.gz](https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_Aml_lede_12.21.0520/openwrt_lede_s905l3a_k5.15.84_2022.12.21.img.gz)，然后修改了`uEnv.txt`的dtb为`meson-gxl-s905l2-x7-5g.dtb`，然后使用`openwrt-install-amlogic`使用`113`进行写入emmc，emmc写入成功，就修改了网络接口为DHCP。
然后发现系统的挂载分区是这样的，不知道是我的刷机操作有误，还是系统的bug，我现在打算手动修复这个挂载错误。
这个挂载应该在**mmcblk2p4**。

```text
root@OpenWrt-S905L3 /mnt [SIGINT]# lsblk

NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
mmcblk2      179:0    0 14.6G  0 disk 
├─mmcblk2p1  179:1    0  160M  0 part /boot
├─mmcblk2p2  179:2    0  960M  0 part /mnt/mmcblk2p4/docker/btrfs
│                                     /mnt/mmcblk2p4/docker
│                                     /
├─mmcblk2p3  179:3    0  960M  0 part /mnt/mmcblk2p3
└─mmcblk2p4  179:4    0 12.3G  0 part 
mmcblk2boot0 179:32   0    4M  1 disk 
mmcblk2boot1 179:64   0    4M  1 disk 
```

[image]

ophub, 2022-12-22:
mmcblk2p2应该挂载到/mnt/mmcblk2
mmcblk2p4应该挂载到/mnt/mmcblk4

sollyu, 2022-12-22:
我是完全用U盘重新刷的。
我的上一条也能看出来，mmcblk2p4是挂载到了`mmcblk2p2`里。
我自己使用`mount -t btrfs /dev/mmcblk2p4 /mnt/mmcblk2p4/`重新挂载了。

ophub, 2022-12-22:
从你的截图反馈里，我找到原因了，因为自定义dtb只留了一个dtb的输入项，所以soc等默认都是空，这个问题我下午修复了：

https://github.com/ophub/luci-app-amlogic/commit/3b88f136e7aa5e0d4af03721277e64db67ef0367

以后在插件里自定义dtb时会同时显示soc和uboot的输入项，这3个选项是很重要的。
在使用脚本安装选择0时也提供了soc等全部输入项的自定义。
这样就解决了你今天碰到的问题了。

closed 2022-12-22 by sollyu

ophub, 2022-12-22:
你的设备在序列里是116，专门为你的设备添加的，不是113，他们2个的dtb相同，但是u-boot不同，根据之前用户反馈使用的是u-boot-p212.bin
把你的这个文件里的值更新下：
UBOOT_OVERLOAD='u-boot-p212.bin'

你方便的话可以测试确认下这个文件，
```
cd /boot
cp -f u-boot-p212.bin u-boot.emmc
chmod +x u-boot.emmc
sync && reboot
```

sollyu, 2022-12-22:
已经测试了，系统正常启动，没有什么异常，还需要其他的额外的测试吗？比如功能、重新用U盘刷写之类的，或者使用`116`再刷一遍？

ophub, 2022-12-22:
不用了，这就可以了。你选择5.15内核的固件，脚本默认不复制u-boot，你手动复制了就已经测试好了。

记得改你本地的那个记录文件啊(你现在手动安装的，需要检查这个值是否正确，方便你以后使用其他内核时不会出错)
/etc/flippy-openwrt-release

SOC='s905l3'
UBOOT_OVERLOAD='u-boot-p212.bin'

sollyu, 2022-12-22:
安装完后`/etc/flippy-openwrt-release`的内容
```shell
PLATFORM='amlogic'
FAMILY='meson-g12a'
BOARD='s905l3a'
KERNEL_VERSION='5.15.84'
K510='0'
FDTFILE='meson-gxl-s905l2-x7-5g.dtb'
UBOOT_OVERLOAD='u-boot-s905x-s912.bin'
MAINLINE_UBOOT=''
ANDROID_UBOOT=''
SOC='s905l2'
```
已经修改成了
```shell
PLATFORM='amlogic'
FAMILY='meson-g12a'
BOARD='s905l3a'
KERNEL_VERSION='5.15.84'
K510='0'
FDTFILE='meson-gxl-s905l2-x7-5g.dtb'
UBOOT_OVERLOAD='u-boot-p212.bin'
MAINLINE_UBOOT=''
ANDROID_UBOOT=''
SOC='s905l3'
```
并且使用`sync && reboot`重启后没有发现异常

ophub, 2022-12-22:
好的，完美。

在系统菜单下---晶晨宝盒---在线下载更新---仅更新宝盒插件，点这个按钮把插件也升级到最新（以后每次更新内核或者固件前，记得都先升级插件到最新，再更新固件或内核）。

sollyu, 2022-12-22:
收到，感谢大佬的耐心指导。

ophub, 2022-12-22:
不客气，没有你的反馈，我都发现不了这个bug的存在，你解决了很大的问题。

sollyu, 2022-12-22:
:tada: 能帮助大佬，真的是荣幸，辛苦大佬了。