ophub/amlogic-s9xxx-openwrt #357: [s905x] B860H can't booting with latest kernel 5.15 with Openwrt Official

Manssizz, 2022-12-24:
Standard chat template, no routine, no chat.
Hi... i just want to report if b860h cant booting with latest kernel 5.15
for now i try to compile all kernel without latest kernel version.

**Device Information**
 - SOC: S905X
 - Model B860H

**Armbian Version**
 - Kernel Version: 5.15.80
 - Release: Official and Lede

**Describe the bug**
booting log showing like 

```
[    6.949353] BTRFS info (device mmcblk1p2): using free space tree
[    6.955281] BTRFS info (device mmcblk1p2): has skinny extents
[    6.995911] BTRFS info (device mmcblk1p2): enabling ssd optimizations
[    7.214083] init: Console is alive
[    7.635606] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/6lowpan.ko
[    7.640583] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/8189fs.ko
[    7.647963] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/8723cs.ko
[    7.655332] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/88x2cs.ko
[    7.662720] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/9p.ko
[    7.669796] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/9pnet.ko
[    7.677118] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/9pnet_virtio.ko
[    7.685060] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/a3d.ko
[    7.692215] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/ac.ko
[    7.699290] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/ac200.ko
[    7.706616] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/acard-ahci.ko
[    7.714431] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/acpi_configfs.ko
[    7.722416] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/act_bpf.ko
[    7.729918] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/act_connmark.ko
[    7.737839] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/act_csum.ko
[    7.745446] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/act_ct.ko
[    7.752831] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/act_ctinfo.ko
[    7.760607] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/act_gact.ko
[    7.768200] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/act_ife.ko
[    7.775701] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/act_ipt.ko
[    7.783189] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/act_meta_mark.ko
[    7.791334] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/act_meta_skbprio.ko
[    7.799538] kmodloader: failed to mmap /lib/modules/5.15.85-flippy-80+o/act_meta_skbtcindex.ko
. . . 
```

ophub, 2022-12-24:
Can the 5.10 kernel boot?

Manssizz, 2022-12-24:
5.10 untested. 
but im curious. i try boot with your lede image, my device can booting. 
maybe i must correct my build config file, then reporting if any problem. so... temporary closed
[image]

closed 2022-12-24 by Manssizz