ophub/amlogic-s9xxx-openwrt #358: [S905x] B860H Wireless Bug

Manssizz, 2022-12-25:
Standard chat template, no routine, no chat.
Tested few kernel
- 5.10 => Kernel Hung
- 5.15.70 + 5.15.85 => Wireless bug

**Device Information | 设备信息**
 - SOC: S095x
 - Model: B860H

**Armbian Version | 系统版本**
 - Kernel Version: 5.15.70 + 5.15.85
 - Release: lede

**Describe the bug | 问题描述**
2 Build i use:
- ophub
- me

both of build have a bug. 
for kernel hung i forget to dump logs.
for wireless bug, logs available [here](https://pastebin.com/raw/z5KcKiYm)

ophub, 2022-12-25:
What is a wireless chip?
Try updating to 5.4 kernel

Manssizz, 2022-12-26:
Realtek rtl8189es.
I not yet try for kernel 5.4. But if i use build official openwrt from your repository, it's work. Booting successful and wireless on.
I prefer use lede build because i some package i need only compatible on lede builds

closed 2022-12-28 by ophub