ophub/amlogic-s9xxx-openwrt #359: Terminal information for Can't locate strict.pm in @INC

gforce1206, 2022-12-29:
[image]
compile with Lede，sudo ./make -s 2048 -b s905l2 -k 5.10.161

ophub, 2022-12-29:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config#1011-required-options-for-openwrt

```
Languages -> Perl
             -> perl-http-date
             -> perlbase-file
             -> perlbase-getopt
             -> perlbase-time
             -> perlbase-unicode
             -> perlbase-utf8
```

gforce1206, 2022-12-29:
thanks，I try try

gforce1206, 2022-12-29:
> You lack `perl`. According to the documentation, select all the required dependencies you lack.
> 
> https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config#1011-required-options-for-openwrt
> 
> ```
> Languages -> Perl
>              -> perl-http-date
>              -> perlbase-file
>              -> perlbase-getopt
>              -> perlbase-time
>              -> perlbase-unicode
>              -> perlbase-utf8
> ```
> 
> [image]

thanks，I try try

closed 2022-12-29 by ophub