ophub/amlogic-s9xxx-openwrt #360: fork后代码改动自动更新吗？

3136083906, 2022-12-30:
Standard chat template, no routine, no chat.
标准聊天模板，无套路不聊天。

**Device Information | 设备信息**
 - SOC: [e.g. S095x3]
 - Model [e.g. HK1]

**Armbian Version | 系统版本**
 - Kernel Version: [e.g. 5.15.100]
 - Release: [e.g. lede]

**Describe the bug | 问题描述**
Describe the problem in detail and attach screenshots if necessary.
详细描述问题，并在必要时附上屏幕截图。

之前fork本项目，会自动跟随本仓库代码更新吗？

ophub, 2022-12-30:
不会，要手动更新

closed 2022-12-30 by ophub

lovelylain, 2022-12-30:
@ophub 为啥你经常修改历史提交记录，fork后时不时把你的最新修改merge进来，结果碰到你两次修改历史记录的情况，第一次rebase搞定了，第二次github认为是两个完全不相干的提交，实在rebase不动了。

ophub, 2022-12-30:
https://github.com/newren/git-filter-repo

我是用的这个在维护git，我从github官方blog了解到的，他们推荐用这个清理无用的git文件。当某个文件现在仓库已经不再使用时（也有一些是改变了目录或者文件名），会从.git里彻底删除，这会让历史上的Commits发生变化。但是这种结果我认可，不再使用的文件留着占用空间，不用了就可以永久删除。

我在使用这个工具之前我的仓库有几个G(以前存放过很长时间的内核包，每套50Mb)，经过清理后总仓库目前不到10Mb了，这样git clone也快了，管理也方便了，优点多多，所以我己经养成了不定期检查清理无用文件的习惯。