ophub/amlogic-s9xxx-openwrt #363: M401A 跑甜糖系统日志出现这个 百度了一下好像是断流什么

yuan6975, 2023-01-05:
Standard chat template, no routine, no chat.

**Device Information | 设备信息**
 - SOC: [e.g. S095l3a]
 - Model [e.g. m401a]

**Armbian Version | 系统版本**
 - Kernel Version: [e.g. 5.15.55]
 - Release: [e.g. lede]

**Describe the bug | 问题描述**

前段时间跑甜糖3个小时左右就会自动关机 
今天把最高频率降到了1700跑了7个小时了  

但是内核日志开始刷新这个错误
[   44.891653] device eth0 left promiscuous mode
[   44.893936] br-lan: port 1(eth0) entered disabled state
[   44.934457] meson8b-dwmac ff3f0000.ethernet eth0: Link is Down
[   46.824531] meson8b-dwmac ff3f0000.ethernet eth0: PHY [0.1:08] driver [Meson G12A Internal PHY] (irq=38)
[   46.837586] meson8b-dwmac ff3f0000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   46.850607] meson8b-dwmac ff3f0000.ethernet eth0: No Safety Features support found
[   46.856429] meson8b-dwmac ff3f0000.ethernet eth0: PTP not supported by HW
[   46.863175] meson8b-dwmac ff3f0000.ethernet eth0: configuring for phy/rmii link mode
[   46.871969] br-lan: port 1(eth0) entered blocking state
[   46.876053] br-lan: port 1(eth0) entered disabled state
[   46.881516] device eth0 entered promiscuous mode
[   48.442965] meson8b-dwmac ff3f0000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   48.450107] br-lan: port 1(eth0) entered blocking state
[   48.455164] br-lan: port 1(eth0) entered forwarding state
[   48.460738] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   68.974351] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   68.979896] NFSD: Using legacy client tracking operations.
[   68.985312] NFSD: starting 10-second grace period (net f0000000)
[  102.500593] sh (17997): drop_caches: 3
[  155.118783] sh (18593): drop_caches: 3
[13621.025275] meson_clk_pll_set_rate: pll did not lock, trying to restore old rate 6000000000
[16988.950901] meson_clk_pll_set_rate: pll did not lock, trying to restore old rate 5592000000
[17716.062650] meson_clk_pll_set_rate: pll did not lock, trying to restore old rate 5592000000
[18106.259649] meson_clk_pll_set_rate: pll did not lock, trying to restore old rate 5592000000
[19157.766427] meson_clk_pll_set_rate: pll did not lock, trying to restore old rate 5592000000
[19378.877493] meson_clk_pll_set_rate: pll did not lock, trying to restore old rate 5592000000
[19485.205861] meson_clk_pll_set_rate: pll did not lock, trying to restore old rate 3408000000
[21511.632069] meson_clk_pll_set_rate: pll did not lock, trying to restore old rate 6000000000
[21511.643928] TCP: request_sock_TCP: Possible SYN flooding on port 56391. Sending cookies.  Check SNMP counters.
[24962.490183] meson_clk_pll_set_rate: pll did not lock, trying to restore old rate 3216000000

大佬有空解答一下吧

yuan6975, 2023-01-05:
盒子版本 是TYPE—A 强体质版本  写入是用的最早的22c的dtb 306 写入的 应该是写入了主线uboot的
root@M401A:~# cat /boot/uEnv.txt
LINUX=/zImage
INITRD=/uInitrd
FDT=/dtb/amlogic/meson-g12a-s905l3a-e900v22c.dtb
APPEND=root=UUID=f355e986-31d5-458b-a6ad-fe25b77bbc14 rootfstype=btrfs rootflags=compress=zstd:6 console=ttyAML0,115200n8 console=tty0 no_console_suspend consoleblank=0 fsck.fix=yes fsck.repair=yes net.ifnames=0 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1

ophub, 2023-01-05:
dtb用m401a的试试，这个降频了

yuan6975, 2023-01-05:
> 

好的  上次看大佬在说dtb后面再加了一个定义100m网卡的内容是不是彻底解决了开机有概率没有ip的问题
我现在只能远程控制盒子  能直接上传一个meson-g12a-s905l3a-m401a.dtb 文件到/boo/dtb文件夹  然后修改uEnv.txt文件重启吗

ophub, 2023-01-05:
可以

yuan6975, 2023-01-05:
> 

除了下载最新的固件能meson-g12a-s905l3a-m401a.dtb提取到这个文件 github哪里能下载到成品吗 谢谢

yuan6975, 2023-01-05:
更换dtb文件以后日志还是出现了
   ]70.632111] NFSD: starting 10-second grace period (net f0000000)
[   94.276863] sh (18189): drop_caches: 3
[ 3769.170966] meson_clk_pll_set_rate: pll did not lock, trying to restore old rate 5592000000
[ 3769.188392] TCP: request_sock_TCP: Possible SYN flooding on port 34340. Sending cookies.  Check SNMP counters.
[ 4211.428350] meson_clk_pll_set_rate: pll did not lock, trying to restore old rate 5592000000
[ 7573.956384] sh (12809): drop_caches: 3