ophub/amlogic-s9xxx-openwrt #373: 最新版报错

vpei, 2023-01-18:
**Device Information | 设备信息**
 - SOC: [S095d]
 - Model [N1]
 
5.15.88 5.15.87 都不行。

amlogic-s9xxx-openwrt/tmp/5.15.88/s905d/rootfs: wrong fs type, bad option, bad superblock on /dev/loop1p2, missing codepage or helper program, or other error. [ ERROR ] mount /dev/loop1p2 failed!

上次更新make后就不行，之前其他文件都没变，都可以用。

ophub, 2023-01-18:
https://github.com/ophub/amlogic-s9xxx-openwrt/actions

我的挺好

vpei, 2023-01-18:
> https://github.com/ophub/amlogic-s9xxx-openwrt/actions
> 
> 我的挺好 你打包的服务器没有安装btrfs的依赖 sudo apt update sudo apt install btrfs-progs

我试试，谢谢。

vpei, 2023-01-18:
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
btrfs-progs 已经是最新版 (4.20.1-2)。
升级了 0 个软件包，新安装了 0 个软件包，要卸载 0 个软件包，有 0 个软件包未被升级。

应该是安装过了，可能是缺少其他软件，前几天好像卸载了一些软件。

ophub, 2023-01-18:
https://github.com/ophub/amlogic-s9xxx-openwrt#local-build-instructions

一键安装全部依赖

vpei, 2023-01-20:
> https://github.com/ophub/amlogic-s9xxx-openwrt#local-build-instructions
> 
> 一键安装全部依赖

我用的系统是debian 10 之前的都可以用，现在不能用，虚拟机新装了系统，按一键安装了依赖项，也不行，只有mkisofs这一项安装不上，以前也是没安装这个。

vpei, 2023-01-20:
Ubuntu上编译成功，不知道是什么原因。我之前一直用debian 10 都是可以的。

closed 2023-01-20 by vpei