ophub/amlogic-s9xxx-openwrt #374: Meson project

esaaprillia, 2023-01-27:
Hy @ophub 

can we make a kernel and boot without using armbian? I saw a project on openwrt for meson it's great if you change your build

https://github.com/openwrt/openwrt/pull/10331

ophub, 2023-01-27:
https://github.com/ophub/amlogic-s9xxx-armbian/tree/main/compile-kernel

The method of compiling and installing the kernel in armbian is detailed in the instructions

```
# Compile the kernel
armbian-kernel -u
armbian-kernel -k 5.15.90

# install the kernel（Execute the command in the kernel file directory）
armbian-update
```

esaaprillia, 2023-01-27:
> https://github.com/ophub/amlogic-s9xxx-armbian/tree/main/compile-kernel
> 
> Metode kompilasi dan penginstalan kernel di armbian dirinci dalam instruksi
> 
> ```
> # Compile the kernel
> armbian-kernel -u
> armbian-kernel -k 5.15.90
> 
> # install the kernel（Execute the command in the kernel file directory）
> armbian-update
> ```

I don't really like your current build using the kernel from armbian. can't you use the method in the Meson project?

esaaprillia, 2023-01-27:
> https://github.com/ophub/amlogic-s9xxx-armbian/tree/main/compile-kernel
>
> The method of compiling and installing the kernel in armbian is detailed in the instructions
>
> ```
> # Compile the kernel [...]
I don't really like your current build using the kernel from armbian. can't you use the method in the Meson project?

ophub, 2023-01-27:
use ubuntu

esaaprillia, 2023-01-27:
> use ubuntu

I don't understand what anyone is saying. currently openwrt manages the kernel itself. but your build uses the armbian kernel and boots.

closed 2023-01-27 by esaaprillia

ophub, 2023-01-27:
Because openwrt does not support these TV boxes, these boxes are usually despised and developers like development boards.
Although he was named amlogic-s9xxx-openwrt, he is not a standard openwrt, but a variant of the armbian startup method.

esaaprillia, 2023-01-27:
> Because openwrt does not support these TV boxes, these boxes are usually despised and developers like development boards.
> 
> Although he was named amlogic-s9xxx-openwrt, he is not a standard openwrt, but a variant of the armbian startup method.
> 
> I feel that this kind of TV box will not be accepted by the mainstream. Like the attitude of many people in your post above, these boxes have been despised, and I can't understand why there is such a group phenomenon. Perhaps like some points mentioned in that post, these boxes are too cheap, and there are too many variants, manufacturers are not open source, users are few, and support is complex, so they are abandoned by the mainstream.
> 
> We are just a kind of entertainment. We are independent of the world and enjoy ourselves.

i will speak regardless of support from openwrt developers. is it possible to manage the kernel in the way openwrt uses? now you build your kernel and boot with armbian

ophub, 2023-01-27:
As you can see from the make script, this openwrt replaces /boot and /lib/modules

So his boot and kernel use armbian's mechanism, which is slightly different from openwrt

https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/make#L436

esaaprillia, 2023-01-27:
> As you can see from the make script, this openwrt replaces /boot and /lib/modules
> 
> So his boot and kernel use armbian's mechanism, which is slightly different from openwrt
> 
> https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/make#L436
> 
> Every system has a standard that has been formed, and the practice of merging armbian seems ridiculous to the openwrt manager.
> 
> This project is just a kind of entertainment, learning and integration, for a small number of TV box users to play, so the free combination of armbian and openwrt, this approach is difficult to be accepted by the public, but it works well.

thank you for your explanation. i was thinking we could make a kernel and boot like the one in openwrt to package the kernel

esaaprillia, 2023-09-29:
@ophub 

I looked again at the meson project. don't you want to try it? https://github.com/openwrt/openwrt/pull/13504