ophub/amlogic-s9xxx-openwrt #377: 关于个性化插件的问题

nhtcool, 2023-02-06:
大佬，看了您的固件编译教程学到很多，有个问题 就是当我需要某个插件的时候，比如ipsec ,然后我在config里面找到多个和这个有关的代码，比如CONFIG_PACKAGE_luci-app-ipsec-server is not set， CONFIG_PACKAGE_luci-app-ipsec-vpnd is not set，我是需要把所有和ipsec有关的代码都改成 =y 吗？但是我怎么确定关于这个插件ipsec还需要安装哪些东西呢。谢谢····

ophub, 2023-02-06:
如果想精准的选择上插件的全部依赖，就要在本地安装ubuntu，通过make menuconfig进行选择。

一般来说这样手改即可，编译时会自动diff加上你漏掉的。等编译完成了，在releases里面，固件列表中第一个那个config就是编译时生成的准确的.config配置文件。

nhtcool, 2023-02-06:
原来是这样 谢谢解惑  看来要学习的还很多

ophub, 2023-02-06:
https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_Aml_lede_02.05.0157

比如这个列表的第一个config文件就是在这次成功编译时生成的最终版。对应的luci-app-xxx在这里：

https://github.com/coolsnowwolf/luci/tree/master/applications

你想了解哪个插件，除了luci-app-xxx里=y，其他的依赖有没有要选择的，点开看Makefile的LUCI_DEPENDS，比如这个广告过滤插件

https://github.com/coolsnowwolf/luci/blob/bfa46977a5f9cf3d525cec4636647c54cf9d1319/applications/luci-app-adblock/Makefile#L7

nhtcool, 2023-02-06:
原理懂了，现在要精准的话看来还需要安装ubuntu,这个还没有用过，所以要学的东西很多。还有个办法就是想要的功能全部通过第三方，就可以不用本地化，不知道这样理解对不对

ophub, 2023-02-06:
恩，其实如果你的网络访问国际出口不怎么稳定，本地编译会无法拉取很多依赖，openwrt的源码里，很多包要从其他网站/仓库下载，这对网络的连通性要求很高。在 github 上使用 Actions 依赖微软的网络环境，都可以正常拉取到，而且 github 提供的免费服务器也不错，如果自己的电脑没有E5强，网络没微软的顺畅，这是不错的选择。

nhtcool, 2023-02-06:
谢谢指导  先一步步尝试了  满足需求就可以了

closed 2023-02-06 by ophub