ophub/amlogic-s9xxx-openwrt #378: Openwrt H96MAX RK3318

yoelwib11, 2023-02-07:
I see that there is a dtb for the a95x rk3318 on new kernel, but i don't see any firmware for that box. I am using h96max box with rk3318. So how do i use openwrt using this box?

Thank you so much.

ophub, 2023-02-07:
need u-boot

yoelwib11, 2023-02-07:
Yes i have. But which firmware should i use?

ophub, 2023-02-07:
https://github.com/ophub/amlogic-s9xxx-armbian/blob/0d6a02d871cf46f8f7b023c59f43eecde76d9ca8/build-armbian/armbian-files/common-files/etc/model_database.conf#L165

u-boot.itb
idbloader.img
rk3318-xxx.dtb

yoelwib11, 2023-02-07:
Yes i have both. So, what should i do? Please tell me how.

ophub, 2023-02-07:
Upload these files, I will make one by hand, and you can test whether it can be used.

yoelwib11, 2023-02-07:
[H96 Max RK3318.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/10673290/H96.Max.RK3318.zip)

ophub, 2023-02-07:
https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_Aml_lede_02.06.2323

download: `openwrt_rk3318_h96max.img.gz`

yoelwib11, 2023-02-07:
Btw the firmware on my box is still an Android TV.

ophub, 2023-02-07:
I made it using the 6.1.10 kernel. I use your 2 u-boots, and the dtb file uses the one with h96max. As long as these three files are correct, it should be able to start normally in theory.

See the link above for how to boot from USB.

yoelwib11, 2023-02-07:
I just downloaded it from https://cloud.mail.ru/public/KGXQ/a5GcEg8t9/Rockchip/LINUX/Armbian_rk3318.rar. 
And then i burn trust.img and uboot.img.

Yes but why your firmware can't be changed the dtb file? I mean unlike amlogic firmware which boot partition is editable in Windows.

ophub, 2023-02-07:
amlogic is in fat32 format, so it can.
This rockchip uses the ext4 format, which does not support opening.

You can mount the modification under ubuntu/armbian and other systems
```
sudo losetup -P -f --show openwrt.img
sudo mount /dev/loopxxxp1 xxxx
sudo mount /dev/loopxxxp2 yyyy
```

yoelwib11, 2023-02-08:
[rk3318-h96max-box.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/10683158/rk3318-h96max-box.zip)

Can you edit the dtb with this one please? Maybe it is because dtb. Now i am boot Armbian with this dtb and kernel 6.1.9-rockchip64 on emmc.

yoelwib11, 2023-02-08:
I just change dtb by myself but stuck in here. Just reboot and reboot. Can you help me please?
[image]

ophub, 2023-02-08:
Armbian_5.99_Rk3328-tv_Debian_buster_default_4.4.154_desktop_20191121.img

kernel: `4.4.154-rk3328-tv`

```
LABEL Armbian
  LINUX /Image
  INITRD /uInitrd
  FDT /dtb/rockchip/rk3328-mx10-fast.dtb
#  FDT /dtb/rockchip/rk3328-a5x.dtb
#  FDT /dtb/rockchip/rk3328-box-trn9.dtb
  APPEND root=LABEL=ROOTFS rootflags=data=writeback rw console=uart8250,mmio32,0xff130000 console=tty0 no_console_suspend consoleblank=0 fsck.fix=yes fsck.repair=yes net.ifnames=0 
```

What is the purpose of the `rk3328` Armbian you provided? What does this have to do with your H96MAX(rk3318)?

yoelwib11, 2023-02-08:
Finally. Thank you so much. I just change base firmware armbian on emmc and flash your openwrt firmware on sdcard, change dtb with wifi on. But it can't show on monitor. Just blank. I don't know why. 
[image]

ophub, 2023-02-10:
Openwrt is configured and managed daily through browser access or ssh access without a monitor.
No HDMI output is not supported for DTB files.

closed 2023-02-10 by ophub

osnipezzini, 2023-08-02:
> Finally. Thank you so much. I just change base firmware armbian on emmc and flash your openwrt firmware on sdcard, change dtb with wifi on. But it can't show on monitor. Just blank. I don't know why. [image]

Can you post how do you can make this work ?

aagym888, 2023-08-21:
> https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_Aml_lede_02.06.2323
> 
> download: `openwrt_rk3318_h96max.img.gz`
> 
> Set the box to boot from USB/TF/SD： https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config#1010-set-the-box-to-boot-from-usbtfsd
> 
> Test whether it can start normally Default ip: `192.168.1.1` Username: `root` Password: `password`
> 
> It is best to connect the device to the `monitor`, and you can see the result by observing the startup code. After the code stops running, press `Enter` on the keyboard to see the `colored banner`

can u send the download link again?