ophub/amlogic-s9xxx-openwrt #384: Enabling Access Point WLAN makes the openwrt kernel panic.

gunawan092w, 2023-02-10:
Standard chat template, no routine, no chat.
标准聊天模板，无套路不聊天。

**Device Information | 设备信息**
 - SOC: [e.g. S095x3] S905X
 - Model [e.g. HK1] HG680P

**Armbian Version | 系统版本**
 - Kernel Version: [e.g. 5.15.100] 5.15.92-flippy-81+0
 - Release: [e.g. lede] Official Aml

**Describe the bug | 问题描述**
Describe the problem in detail and attach screenshots if necessary.
详细描述问题，并在必要时附上屏幕截图。

Hello, i've been having problem for past few hours. when the AML-OpenWRT Version released today, i was gonna try that firmware and hoping that the problem will be fixed. but instead, it didn't. so i have to make issue on here.

LOGS:
```
C:\Users\Gunawan092w>ssh -l root 192.168.1.1
      ____                 _       __     __        ____
     / __ \____  ___  ____| |     / /____/ /_      / __ )____  _  __
    / / / / __ \/ _ \/ __ \ | /| / / ___/ __/_____/ __  / __ \| |/_/
   / /_/ / /_/ /  __/ / / / |/ |/ / /  / /_/_____/ /_/ / /_/ />  <
   \____/ .___/\___/_/ /_/|__/|__/_/   \__/     /_____/\____/_/|_|
       /_/  H E L L O - W O R L D    W I R E L E S S - F R E E D O M
───────────────────────────────────────────────────────────────────────
 Install OpenWrt: System → Amlogic Service → Install OpenWrt
 Update  OpenWrt: System → Amlogic Service → Online  Update
 Board: Amlogic s905x | OpenWrt Kernel: 5.15.92-flippy-81+o
 Production Date: 2023-02-10
───────────────────────────────────────────────────────────────────────
 Device Model: Amlogic Meson GXL (S905X) P212 Development Board
 Architecture: AArch64 : Cortex-A53 x 4
 Load Average: 0.44 0.21 0.08   Uptime: 1m 24s
 Ambient Temp: 51.0 °C          CPU Freq: 1512 Mhz
 Memory Usage: 8% of 1810M      Swap Usage: 0% of 0M
 Boot Storage: 40% of 251.0M    ROOTFS: 32% of 1.0G
───────────────────────────────────────────────────────────────────────
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# logread -f
Fri Feb 10 06:03:53 2023 kern.warn kernel: [   88.694727] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:03:53 2023 kern.warn kernel: [   88.699639] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:03:53 2023 kern.warn kernel: [   88.708828] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:03:53 2023 kern.warn kernel: [   88.716743] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:03:53 2023 kern.warn kernel: [   88.719772] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:03:58 2023 kern.warn kernel: [   93.724452] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:03:58 2023 kern.warn kernel: [   93.728665] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:03:58 2023 kern.warn kernel: [   93.737157] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:03:58 2023 kern.warn kernel: [   93.742095] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:03:58 2023 kern.warn kernel: [   93.744804] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:04:03 2023 kern.warn kernel: [   98.737563] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:04:03 2023 kern.warn kernel: [   98.740981] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:04:03 2023 kern.warn kernel: [   98.748633] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:04:03 2023 kern.warn kernel: [   98.753244] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:04:03 2023 kern.warn kernel: [   98.756441] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:04:08 2023 kern.warn kernel: [  103.708406] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:04:08 2023 kern.warn kernel: [  103.711638] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:04:08 2023 kern.warn kernel: [  103.719298] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:04:08 2023 kern.warn kernel: [  103.725332] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:04:08 2023 kern.warn kernel: [  103.727748] RTL871X: cfg80211_rtw_get_txpower
Fri Feb 10 06:04:10 2023 daemon.notice netifd: radio0 (5175): ./mac80211.sh: eval: line 1164: ipconfig: not found
Fri Feb 10 06:04:10 2023 daemon.notice netifd: radio0 (5175): ./mac80211.sh: eval: line 1075: ipconfig: not found
Fri Feb 10 06:04:10 2023 daemon.notice netifd: radio0 (5175): ./mac80211.sh: eval: line 1205: ipconfig: not found
Fri Feb 10 06:04:11 2023 kern.info kernel: [  106.385753] rtl8189fs mmc0:0001:1 phy0-ap0: renamed from wlan0
Fri Feb 10 06:04:11 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy phy0-ap0) --> new PHY
[... 78 lines cut ...]
Fri Feb 10 06:04:12 2023 kern.alert kernel: [  107.807435]   SET = 0, FnV = 0
Fri Feb 10 06:04:12 2023 kern.alert kernel: [  107.810519]   EA = 0, S1PTW = 0
Fri Feb 10 06:04:12 2023 kern.alert kernel: [  107.813562]   FSC = 0x21: alignment fault
Fri Feb 10 06:04:12 2023 kern.alert kernel: [  107.817527] Data abort info:
Fri Feb 10 06:04:12 2023 kern.alert kernel: [  107.820369]   ISV = 0, ISS = 0x00000021
Fri Feb 10 06:04:12 2023 kern.alert kernel: [  107.824170]   CM = 0, WnR = 0
Fri Feb 10 06:04:12 2023 kern.alert kernel: [  107.827100] swapper pgtable: 4k pages, 39-bit VAs, pgdp=0000000002421000
Fri Feb 10 06:04:12 2023 kern.alert kernel: [  107.833752] [ffffffc0084e6044] pgd=1000000077fff003, p4d=1000000077fff003, pud=1000000077fff003, pmd=10000000036c2003, pte=0068000007b85703
Fri Feb 10 06:04:12 2023 kern.emerg kernel: [  107.846857] Internal error: Oops: 96000021 [#1] PREEMPT SMP
client_loop: send disconnect: Connection reset by peer
```

Video:
https://drive.google.com/file/d/124D4pxZmDwLMMBK9HOxNhnl5bgRfwtb0/view?usp=drivesdk

Specs:
Device Model: Amlogic Meson GXL (S905X) P212 Development Board
Architecture: AArch64
CPU: Coretex-A53
Storage: 8GB
RAM: 2GB
WIFI: MAC80211 802.11b/g??

Screenshot:
[image]

ophub, 2023-02-10:
Wireless not available, disable wireless, reboot.

gunawan092w, 2023-02-10:
oh. okay thanks

gunawan092w, 2023-02-10:
but wait
why does aml-openwrt's wireless is not available?

ophub, 2023-02-11:
Drive instability

closed 2023-02-11 by ophub