ophub/amlogic-s9xxx-openwrt #390: s905w

chuanshuojibi, 2023-02-19:
Standard chat template, no routine, no chat.
标准聊天模板，无套路不聊天。

**Device Information | 设备信息**
 - SOC: [e.g. S095xw]
 - Model [e.g. x96mini]

**Armbian Version | 系统版本**
 - Kernel Version: [e.g. 5.15.100]
 - Release: [e.g. immortal]

**Describe the bug | 问题描述**
Describe the problem in detail and attach screenshots if necessary.
详细描述问题，并在必要时附上屏幕截图。
[image]
as far as I know my amlogic s905w ‘s  main frequency is about 2Ghz ,while the firmware seems to limit it to 1.2ghz . i wonder is it because higer frequency will make it unstable or something else, and could we make it high to it's normal .also ,can the firmware support a usb3.0 to 1000mbs port?
[image]
[image]
话说这s905w应该是2.0ghz的 怎么只能调度到1.2g，是因为温度过高吗？我想知道能让它提升到2.0ghz 还有这个设备的usb3.0接口能否转接为千兆网卡，我想知道驱动是否支持。

ophub, 2023-02-19:
频率可以在dts里设置，x96-mini默认使用的通用配置，没有特殊限制。

openwrt默认启用的敏捷模式，在需要高性能时会自动刷新到高频，在dts里设置的频率直接自动调节。

https://github.com/unifreq/linux-5.15.y/blob/main/arch/arm64/boot/dts/amlogic/meson-gxl-s905w-x96-mini.dts

有的百兆盒子，在dts里明确限制了网卡百兆，所以使用千兆的USB网卡接USB3.0也会被限速。具体得看dts里的设置，你通过上面的链接，在内核库里查看。

chuanshuojibi, 2023-02-19:
多谢了

closed 2023-02-19 by ophub