ophub/amlogic-s9xxx-openwrt #395: OpenWrt_lede_02.25.1126 无法设置默认的密码 无法理解

abirvepete, 2023-02-25:
如题，电脑连接m401a后进入192.168.1.1之后，系统提醒没有设置密码，跳转到/admin/system/admin设置用户密码。无法设置
为什么会有这么无语的步骤？非要连一次电视？

ophub, 2023-02-26:
哪个源码的没有密码？

abirvepete, 2023-02-26:
哦 我是凌晨1点左右下载的最新的lede的。OpenWrt_lede_02.25.1126。用的是s905l3a那个镜像。已经修改dtb，插好u盘启动进入192.168.1.1。password这个密码登陆失败，提示的设置密码界面根本没有显示。我是盒子与电脑直连的，无外网。获取Outlook for Android

abirvepete, 2023-02-26:
哦 我是凌晨1点左右下载的最新的lede的。OpenWrt_lede_02.25.1126。用的是s905l3a那个镜像。已经修改dtb，插好u盘启动进入192.168.1.1。password这个密码登陆失败，提示的设置密码界面根本没有显示。我是盒子与电脑直连的，无外网。发件人: ***@***.*** ***@***.***> 代表 ophub ***@***.***>发送时间: 2023年2月26日星期日 上午8:01收件人: amlogic-s9xxx-openwrt ***@***.***>抄送: 2597572922 ***@***.***>; author ***@***.***>主题: Re: [ophub/amlogic-s9xxx-openwrt] OpenWrt_lede_02.25.1126 无法设置默认的密码 无法理解 (Issue #395)

哪个源码的没有密码？

abirvepete, 2023-02-26:
哦 我是凌晨1点左右下载的最新的lede的。OpenWrt_lede_02.25.1126。用的是s905l3a那个镜像。已经修改dtb，插好u盘启动进入192.168.1.1。password这个密码登陆失败，提示的设置密码界面根本没有显示。我是盒子与电脑直连的，无外网。发件人: ***@***.*** ***@***.***> 代表 ophub ***@***.***>发送时间: 2023年2月26日星期日 上午8:01收件人: amlogic-s9xxx-openwrt ***@***.***>抄送: 2597572922 ***@***.***>; author ***@***.***>主题: Re: [ophub/amlogic-s9xxx-openwrt] OpenWrt_lede_02.25.1126 无法设置默认的密码 无法理解 (Issue #395)哪个源码的没有密码？

abirvepete, 2023-02-26:
哦 我是凌晨1点左右下载的最新的lede的OpenWrt_lede_02.25.1126。用的是s905l3a那个镜像。已经修改dtb，插好u盘启动进入192.168.1.1。password这个密码登陆失败，提示的设置密码界面根本没有显示。我是盒子与电脑直连的，无外网。发件人: ***@***.*** ***@***.***> 代表 ophub ***@***.***>发送时间: 2023年2月26日星期日 上午8:01收件人: amlogic-s9xxx-openwrt ***@***.***>抄送: 2597572922 ***@***.***>; author ***@***.***>主题: Re: [ophub/amlogic-s9xxx-openwrt] OpenWrt_lede_02.25.1126 无法设置默认的密码 无法理解 (Issue #395)哪个源码的没有密码？

ophub, 2023-02-26:
lede源码应该是默认用户root，密码password，上游改了吗？我今天测试一下

abirvepete, 2023-02-26:
很神奇。早上也顺便试了下official版本，是正常的。剩下的镜像我没有试。lede的排版很喜欢。

ophub, 2023-02-26:
你重新下lede再试一试，这个一直都可以。我看了他最近几周的commit记录，没改过密码的设置。

abirvepete, 2023-02-26:
好的。我会。重新再来一次。成功了的话。就自己关闭了PeteAbirve

abirvepete, 2023-02-26:
大概率是制作u盘掉文件了。joker了

closed 2023-02-26 by abirvepete