ophub/amlogic-s9xxx-openwrt #396: opkg问题，list无法更新，手动删除opkg.list里面的文件会还原成最开始的模板。

abirvepete, 2023-02-26:
设备m401a。镜像lede的最新版本。
问题，起初是软件包每次都会说是24小时未更新。后来在安装openclash的时候直接说是没有找到在镜像网站上libcap2.66.1的ipk，我直接翻镜像网站发现版本已经是2.67了。我仔细排查之后，发现opkg update的时候，opkg.list里面的文件压根就没有更新。删除文件之后，再次运行opkg update，不是获得最终的，而是还原了我删掉的，而且我怎么看opkg.list里面有一个文件已经是2021年的了。并不是太懂openwrt的结构，这是被软件包给控制了吗。虽然说手动替换应该okay，但还是有点太麻烦了。

ophub, 2023-02-26:
你安装的xxx软件，在上游openwrt源码里没有，所以opkg更新不了。这些你从第三方添加的软件，去作者的仓库下载最新的ipk更新即可。

abirvepete, 2023-02-26:
哦哦，也就是lede的自带源设计的时候就是专门让它保存不动吗？请教一下，为什么本地有的依赖，那些软件还是会下一遍，但是镜像上面的依赖已经更新了，旧的依赖ipk又已经删了，就卡死在那了。我或许该尝试一下编译镜像了。更奇怪的是，我在lede大佬的镜像上也找到了软件所需要的依赖。软件安装时也是在lede的镜像上寻找，明明有还没找到。太奇怪了
[image]

[image]

[image]

abirvepete, 2023-02-26:
[image]

ophub, 2023-02-26:
一般情况下，不要做删除插件的操作，删除插件的时候会连带删除他的依赖，这些依赖其他插件可能也在用，你的删除操作会导致不可预知的问题。

你想更新新的版本，直接上传新的点更新就可以了，这样不会删除了依赖，会一起更新。

你不要在帖子和图片里显示你的xxx插件，这个帖子一会我会删除了，我不懂你说的软件，我只能告诉你openwrt里插件通用的更新方法。

abirvepete, 2023-02-26:
好的

ophub, 2023-02-26:
在上游openwrt源码里带的插件，软件包，依赖等，都可以使用上游的软件源更新。

第三方的插件，直接下载他的最新版更新即可，不要删除，删除是危险操作，一起删除的依赖可能导致系统崩溃。你只要下载最新版插件更新即可。

有时人们安装了一个高大全的openwrt，会觉得一些插件无用就去一通卸载，这样卸载插件时也卸载了很多依赖，可能会崩溃了系统。

ophub, 2023-02-26:
5分钟后我删除这个帖子，你懂的，不适合讨论。