ophub/amlogic-s9xxx-openwrt #397: 升级遇到报错umount: : no mount point specified.

QQ-War, 2023-03-07:
**Device Information | 设备信息**
 - SOC: S912
 - Model T1

**Armbian Version | 系统版本**
 - Kernel Version: 6.1.15
 - Release: Lede, Linux OpenWrt 6.1.15-flippy-82+ #47 SMP Sat Mar  4 14:01:49 CST 2023 aarch64 GNU/Linux

**Describe the bug | 问题描述**
使用openwrt-update-amlogic升级，遇到报错，如下：
```
Whether to backup and restore the current config files? y/n [n]
NEW_ROOT_NAME: [ mmcblk2p2 ]
NEW_ROOT_MP: [  ]
The loopdev is [ /dev/loop0 ], wait [ 3 ] seconds.
Mount [ /dev/loop0p1 ] -> [ /mnt/mmcblk2p4/boot ] ...
Mount [ /dev/loop0p2 ] -> [ /mnt/mmcblk2p4/root ] ...
FDT Value [ /dtb/amlogic/meson-gxm-phicomm-t1.dtb ]
K510 [ 1 ]
umount [  ]
umount: : no mount point specified.
Umount [  ] failed, Please restart and try again!
```
怀疑是挂载点的问题。
当前的挂载点如下：
```
Filesystem                Size      Used Available Use% Mounted on
udev                    512.0K         0    512.0K   0% /dev
tmpfs                   192.3M    116.0K    192.2M   0% /run
/dev/mmcblk2p3          960.0M    424.3M    443.2M  49% /
tmpfs                   961.5M     23.8M    937.7M   2% /tmp
tmpfs                   512.0K         0    512.0K   0% /dev
cgroup                  961.5M         0    961.5M   0% /sys/fs/cgroup
/dev/mmcblk2p3          960.0M    424.3M    443.2M  49% /mnt/sda4/docker
/dev/mmcblk2p3          960.0M    424.3M    443.2M  49% /mnt/sda4/docker/btrfs
/dev/mmcblk2p1          511.0M    113.7M    397.3M  22% /boot
/dev/mmcblk2p4           11.9G      1.3G     10.6G  11% /mnt/mmcblk2p4
```
从web界面上面看到mmcblk2的p2和p3都是root，但是p2没有挂载，怀疑是不是跟这个有关系？如果是的话，p2应该怎么挂载呢？
```
UUID | 卷标 | 成员 | 用量 |  
-- | -- | -- | -- | --
306b70aa-bfcb-4680-b63c-ae983395cf9f | EMMC_ROOTFS1 | /dev/mmcblk2p2 | 363.21 MB |  
373f1044-3af3-4a61-9fca-6efeff1401db | EMMC_SHARED | /dev/mmcblk2p4 | 1.31 GB |  
4bf2ae24-3f93-42d2-9cd6-3d61484ba81e | EMMC_ROOTFS2 | /dev/mmcblk2p3 | 420.83 MB |  
```

之前在web的挂载点，全局设置这里，点了一下生成设置，然后这个全局设置这里也不正常。
我想要了解：
1. 这个系统的挂载点默认应该是怎么样的，朋友们方便可以贴一下你们的挂载情况。
2. p2和p3的卷标都是ROOTFS，这两个是什么关系。
3. 目前我也没办法通过U盘启动重装，电视上面看要点击按键不让自动启动，但是我只有一个U口需要插U盘。
4. 这个全局设置点击生成设置，是不是失败了？

谢谢。

ophub, 2023-03-07:
```
# 你的op系统当前运行在p3，但是p2有问题，所以先格式化下p2
mkfs.btrfs  /dev/mmcblk2p2 -f
# 格式化完成后挂载下
mount /dev/mmcblk2p2 /mnt/mmcblk2p2
# 保存数据至emmc
sync
# 再次升级
openwrt-update-amlogic
```

QQ-War, 2023-03-07:
好了，感谢。
我理解了，实际上P2是overlay到P3上面去的,P3就是原版的rootfs，然后P2是可以修改的部分，其他像P4和P1都不是系统正常运行一定要挂载的。
这个报错，是因为P2没有挂载。
再进一步是因为全局设置这里，点了一下生成设置，这个操作不知道怎么出了问题，让P3下面的fstab出问题了，同步P2下面的fstab也不对，就导致异常。

closed 2023-03-07 by QQ-War

ophub, 2023-03-07:
p2/p3分区在固件更新时交替使用，每次会更新至另外一个分区，当前分区在更新结束并重启后处于闲置状态。

不要点系统菜单————挂载点——-刷新按扭。openwrt会改了fstab

QQ-War, 2023-03-07:
还有一个问题，就是这个机器插u盘，再上电，还是进入内置的系统，怎么样设置可以重启进入u盘的系统呢？

ophub, 2023-03-07:
方法有多种。

最暴力简单的方法，是你关机前，先把emmc里面系统的/boot/boot.src 等文件改个名字。这样emmc无法启动了，必然从usb启动。

usb如果只是看看，并𣎴用，就在usb里面把emmc挂载到usb里面改回原名。如果是要重装系统，直接重装就覆盖了。

QQ-War, 2023-03-07:
懂了，多谢。