ophub/amlogic-s9xxx-openwrt #401: 使用immortalwrt的源码编译，打包的时候出现错误

gforce1206, 2023-03-12:
**Device Information | 设备信息**
 - SOC: [e.g. S095l3b]
 - Model [e.g. cm211-1]

**Armbian Version | 系统版本**
 - Kernel Version: [e.g. 5.10.172]
 - Release: [e.g. immortalwrt openwrt-21.02 --single-branch]

**Describe the bug | 问题描述**
开发者大佬，麻烦你看看
`[ STEPS ] Start making OpenWrt firmware...
(1.1) Start making OpenWrt [ s905l2 - stable/5.4.234 ]. Remaining space is 54G.
 [ s905l2 - 5.4.234 ]  (1/6) Make OpenWrt image.
 [ s905l2 - 5.4.234 ]  (2/6) Extract OpenWrt files.
mount: /home/book/amlogic-s9xxx-openwrt/tmp/5.4.234/s905l2/rootfs: wrong fs type, bad option, bad superblock on /dev/loop12p2, missing codepage or helper program, or other error.
[ ERROR ] mount /dev/loop12p2 failed!`

ophub, 2023-03-13:
偶尔会这样服务器挂载不上镜像，重打包即可

gforce1206, 2023-03-13:
> 偶尔会这样服务器挂载不上镜像，重打包即可

可以了，谢谢

closed 2023-03-13 by gforce1206