ophub/amlogic-s9xxx-openwrt #408: MGV2000/S905L2/九联（JL）USB启动到一半卡住

stevenni, 2023-03-24:
Standard chat template, no routine, no chat.
标准聊天模板，无套路不聊天。

Device Information | 设备信息

SOC: [S095L3]
Model [MGV2000(JL)]
Armbian Version | 系统版本

Kernel Version: [5.1.103]
Release: [ jammy]
Describe the bug | 问题描述
Describe the problem in detail and attach screenshots if necessary.
详细描述问题，并在必要时附上屏幕截图。

使用USB启动，到一半卡主
1.openwrt 
卡在这个地方，插拔USB也不行
[image]

2.LEDE
卡在这个地方
[image]

ophub, 2023-03-24:
这不叫看住，这叫正常启动吧？
你接个键盘，回车就看到banner了。
或者你把电脑改到192.168.1.x（x不是1的其他数字），浏览器访问192.168.1.1

stevenni, 2023-03-24:
需要回车一下吗，好的我试试，不过现在已经装了armbian了，不知道怎么再装openwrt，第一次玩。另外就是因为电视屏幕分辨率的问题，字符界面前面有1-2列，下面有一行左右看不到。后面装armbian的时候才发现在电视上可以调整，就可以显示完整了。不过装openwrt和LEDE的时候，我都用电脑配置了IP地址ping过192.168.1.1，ping不通。

ophub, 2023-03-24:
直接U盘刷新的系统，armbian或者openwrt，然后断电，插入u盘，插入电源启动，即可安装新系统。

stevenni, 2023-03-24:
好的，因为之前做过软件测试的工作，问题不一样，所以就分别提了不同的问题，后续都在这个帖子里面交流。
IP这个还是懂的，我是搞网络的。5.1.103内核的配置了相同网段的IP的，openwrt和armbian都ping不通。armbian获取不到IP，手动配置了IP（ip addr add 192.168.1.x/24 dev eth0）也是不通的。网卡ip link set up/down也尝试了，不通。

另外6内核的是openwrt和armbian两个系统从usb启动时，都没有到字符界面，直接自动重启了。

stevenni, 2023-03-24:
MGV2000 九联代工（JL），南传版本电路板：
[image]
[image]

closed 2023-03-28 by ophub