ophub/amlogic-s9xxx-openwrt #413: 用github action编译的时候如何修改默认的dtb

ysuolmai, 2023-04-08:
我按照文档的说明，去写-b 参数为如"s905x3-tx3" 或者s905l3a-cm311，编译就没法完成，貌似只能写 s905x3, 或者s905l3a, 那么问题就是有什么办法可以指定写入uEnv.txt的默认dtb么？

ophub, 2023-04-08:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/make-openwrt/documents/README.cn.md

看文档第5章节

其中 BUILD 的值是 yes 的才可以设置为 -b 的指定参数。
s905x3-tx3 是 no

ophub, 2023-04-08:
在本地编译时通过 `-b` 参数指定，在 github.com 的 Actions 里编译时通过 `openwrt_board` 参数指定。使用 `-b all` 代表打包 `BUILD` 是 `yes` 的全部设备。使用指定 `BOARD` 参数打包时，无论 `BUILD` 是 `yes` 或者 `no` 均可打包，例如：`-b r68s_s905x3-tx3_s905l3a-cm311`

https://github.com/ophub/amlogic-s9xxx-openwrt/commit/8588018fa7d7b93e42e83822fc0c1451f1f1e215
我升级了下，当使用 -b 指定具体设备打包时，忽略 BUILD 的设置。

ysuolmai, 2023-04-08:
感谢大佬，现在就编译一下试试

closed 2023-04-11 by ophub

ysuolmai, 2023-04-12:
忘记反馈了不好意思，试过了很完美，现在可以指定给CM311-1A直接编译了