ophub/amlogic-s9xxx-openwrt #415: rk3588 openwrt 开不了机

gwx11111111, 2023-04-13:
HI，ophub

       现在我在用amlogic-s9xxx-openwrt的代码，但是烧录了发现开不了机呢？
日志如下：

DDR Version V1.09 20220721
LPDDR4X, 1584MHz
channel[0] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
 BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[2] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[3] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
Manufacturer ID:0x6
change to F1: 528MHz
change to F2: 1068MHz
change to F3: 1560MHz
change to F0: 1584MHz
h
vv3kFK6v&&6V&ailed to set cpub01
Failed to set cpub23
unknown raw ID phN
unrecognized JEDEC id bytes: 00, 00, 00
Trying to boot from MMC2
spl: partition error
Trying fit image at 0x4000 sector
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(806278dba1...) + OK
## Checking uboot 0x00200000 ... sha256(d37588cc6f...) + OK
## Checking fdt 0x0030a190 ... sha256(d32827d3ac...) + OK
## Checking atf-2 0x000f0000 ... sha256(c00c7fd75b...) + OK
## Checking atf-3 0xff100000 ... sha256(71c3a5841b...) + OK
## Checking atf-4 0xff001000 ... sha256(2301cf73be...) + OK
Jumping to U-Boot(0x00200000) via ARM Trusted Firmware(0x00040000)
Total: 348.738 ms

INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-405-gb52c2eadd:derrick.huang
NOTICE:  BL31: Built : 11:23:47, Aug 15 2022
INFO:    spec: 0x1
INFO:    ext 32k is not valid
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    system boots from cpu-hwid-4
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
INFO:    dfs DDR fsp_params[0].freq_mhz= 1584MHz
INFO:    dfs DDR fsp_params[1].freq_mhz= 528MHz
INFO:    dfs DDR fsp_params[2].freq_mhz= 1068MHz
INFO:    dfs DDR fsp_params[3].freq_mhz= 1560MHz
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

U-Boot 2017.09-g73cc10cb06-220414 #root (Oct 19 2022 - 16:15:35 +0800)

Model: Radxa ROCK 5B
PreSerial: 2, raw, 0xfeb50000
DRAM:  7.7 GiB
Sysmem: init
Relocation Offset: edb6d000
Relocation fdt: eb9f9e58 - eb9fecc8
CR: M/C/I
Using default environment

no mmc device at slot 1
mmc@fe2c0000: 1 (SD), mmc@fe2e0000: 0
Bootdev(atags): mmc 1
MMC1: Legacy, 52Mhz
PartType: EFI
DM: v2
No misc partition
boot mode: None
FIT: No boot partition
No resource partition
No resource partition
No resource partition
Failed to load DTB, ret=-19
No find valid DTB, ret=-22
Failed to get kernel dtb, ret=-22
In:    serial
Out:   serial
Err:   serial
Model: Radxa ROCK 5B
CLK: (sync kernel. arm: enter 1008000 KHz, init 1008000 KHz, kernel 0N/A)
  b0pll 24000 KHz
  b1pll 24000 KHz
  lpll 24000 KHz
  v0pll 24000 KHz
  aupll 24000 KHz
  cpll 1500000 KHz
  gpll 1188000 KHz
  npll 24000 KHz
  ppll 1100000 KHz
  aclk_center_root 702000 KHz
  pclk_center_root 100000 KHz
  hclk_center_root 396000 KHz
  aclk_center_low_root 500000 KHz
  aclk_top_root 750000 KHz
  pclk_top_root 100000 KHz
  aclk_low_top_root 396000 KHz
No misc partition
Net:   No ethernet found.
Hit key to stop autoboot('CTRL+C'):  0 
switch to partitions #0, OK
mmc1 is current device
** Invalid partition 1 **
switch to partitions #0, OK
mmc0(part 0) is current device
Failed to mount ext2 filesystem...
** Unrecognized filesystem type **
starting USB...
No working controllers found
USB is stopped. Please issue 'usb start' first.
这是为什么呢？

ophub, 2023-04-13:
什么机器

gwx11111111, 2023-04-13:
我们机器是:ipc-r
 soc：rk3588
内存：lpddr4 8g
支持多个以太网口（8168或者8125）
机器图片如下链接：

 
[image]

gwx11111111, 2023-04-13:
这个能不能帮忙加一下这个盒子的支持到openwrt里面呢？我们
是beelink的，需要什么资料我这边可以提供的，谢谢！

gwx11111111, 2023-04-13:
HI,ophub

我们机器是:Beelink ipc-r
soc：rk3588
内存：lpddr4 8g
支持多个以太网口（8168或者8125）
机器图片如下

这个能不能帮忙加一下这个盒子的支持到openwrt里面呢？
需要什么资料我这边可以提供的，谢谢！

ophub, 2023-04-13:
目前的RK3588只有 Radxa-Rock5B, HinLink-H88K 这2个设备支持，每个设备需要单独的u-boot和dtb，你有这几个文件没？或者u-boot编译源码？

u-boot.itb和idbloader.img，或者bootloader.bin这样与你的机器适配的。

gwx11111111, 2023-04-13:
我们u-boot源码和dtb都有的，我们要怎么提供给你们呢

gwx11111111, 2023-04-13:
HI，ophub

uboot源码已经推送gittub了地址：https://github.com/gwx11111111/uboot_ipc.git，dtb在邮件附件

，看看这样可以吗?

ophub, 2023-04-13:
https://github.com/unifreq/linux-5.10.y-rk35xx/tree/main/arch/arm64/boot/dts/rockchip

这个dts文件有吗？是5.10内核的？

gwx11111111, 2023-04-13:
HI,ophub
有的，内核（包括dts）已经推上gittub，地址：https://github.com/gwx11111111/kernel_IPC.git
是5.10是内核;

 
https://github.com/unifreq/linux-5.10.y-rk35xx/tree/main/arch/arm64/boot/dts/rockchip

ophub, 2023-04-13:
你这仓库里3588还挺丰富啊，你这个板子用的dts文件名称是哪个？

gwx11111111, 2023-04-13:
hI，ophub
     rk3588的还行吧！板子用的是rk3588-nvr-demo-v10-android.dts

ophub, 2023-04-13:
你的uboot_ipc源码无法编译，第一次编译提示少rkbin仓库。

```
Environment offset (ENV_OFFSET) [0x0] (NEW) 
Environment size (ENV_SIZE) [0x8000] 0x8000
Environment redundant(backup) offset (ENV_OFFSET_REDUND) [0x0] (NEW) 
Environment offset in Nand (ENV_NAND_OFFSET) [0x0] (NEW) 
Environment redundant(backup) offset in Nand (ENV_NAND_OFFSET_REDUND) [0x0] (NEW) 
Environment size in Nand (ENV_NAND_SIZE) [0x40000] (NEW) 
Environment offset in Nor (ENV_NOR_OFFSET) [0x0] (NEW) 
Environment redundant(backup) offset in Nor (ENV_NOR_OFFSET_REDUND) [0x0] (NEW) 
Environment size in Nor (ENV_NOR_SIZE) [0x10000] (NEW) 
#
# configuration written to .config
#
ERROR: No ../rkbin repository
```

我拉了radxa/rkbin 试了下也不行

```
Default Configuration: 'conf'
 Configuration 0 (conf)
  Description:  rk3588-evb
  Kernel:       unavailable
  Firmware:     atf-1
  FDT:          fdt
  Loadables:    uboot
                atf-2
                atf-3
                optee
out:rk3588_download_v1.08.111.bin
E: [parseOpts_from_file] unknown sec: IDB_PATH=rk3588_idblock_v1.08.111.img!
merge failed!
```

你完整的几个源码仓库有吗？你自己编译成功了没？

gwx11111111, 2023-04-13:
HI,ophub
&nbsp; 我这边可以编译的，我没用的是https://github.com/gwx11111111/ipc_rkbin.git
这个厂库的rkbin(bin/rk35/rk3588_ddr_lp4_2112MHz_lp5_2736MHz_v1.08.bin)

gwx11111111, 2023-04-13:
HI,ophub
  我这边可以编译的，我没用的是https://github.com/gwx11111111/ipc_rkbin.git
这个厂库的rkbin(bin/rk35/rk3588_ddr_lp4_2112MHz_lp5_2736MHz_v1.08.bin)

gwx11111111, 2023-04-13:
hi,ophub

还有用这个rk3588_defconfig配置文件

gwx11111111, 2023-04-13:
HI,ophub

  现在可以编译了吗？如果不行的话，这个idbloader.img和rk3588_spl_loader_v1.08.111.bin和 u-boot.itb这3个
要怎么编译出来呢？因为我现在只是编译出uboot.img来用的

gwx11111111, 2023-04-13:
HI,ophub
这是我这边编译好的uboot文件，dtb和dts文件
[rockchip.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/11219375/rockchip.zip)
[uboot.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/11219395/uboot.zip)

ophub, 2023-04-13:
https://github.com/gwx11111111/uboot_ipc/tree/beelink_RK3358_rk12/arch/arm/dts

[image]

ophub, 2023-04-13:
rk3588-nvr-demo-v10-android.dtb
这是你设备的dtb文件？

gwx11111111, 2023-04-13:
> https://github.com/gwx11111111/uboot_ipc/tree/beelink_RK3358_rk12/arch/arm/dts
>
>
>
> 你上面发的这个 rockchip.zip 里的dts文件，如果和你的源码是一套，要放进来，不然没法编译。

uboot直接

有这些就可以了的，rockchip.zip是内核的dts（dtb）

gwx11111111, 2023-04-13:
rk3588-nvr-demo-v10-android.dtb是设备的dtb

gwx11111111, 2023-04-13:
uboot有这些
[image]
已经可以编译里的，就是要rkbin仓库,最后
[image]
用这个rk3588_defconfig配置文件编译

ophub, 2023-04-14:
`u-boot$ ./make.sh rk3588`

[image]

[image]

只用你的2个仓库编译过了，后面编译idbloader.img和u-boot.itb失败了。

你看下 https://wiki.radxa.com/Rock5/guide/build-u-boot-on-5b

我用他的build脚本，改了下 board_configs.sh

```
	"rk3588-ipc")
		DEFCONFIG=rockchip_linux_defconfig
		UBOOT_DEFCONFIG=rk3588_defconfig
		DTB=rk3588-nvr-demo-v10-android.dtb
		export ARCH=arm64
		export CROSS_COMPILE=aarch64-linux-gnu-
		CHIP="rk3588"
		;;
```

改了下 mk-uboot.sh

```
elif [ "${CHIP}" == "rk3588s" ] || [ "${CHIP}" == "rk3588" ]; then
	make ${UBOOT_DEFCONFIG}
	make BL31=../rkbin/bin/rk35/rk3588_bl31_v1.30.elf spl/u-boot-spl.bin u-boot.dtb u-boot.itb
	./tools/mkimage -n rk3588 -T rksd -d ../rkbin/bin/rk35/rk3588_ddr_lp4_2112MHz_lp5_2736MHz_v1.08.bin:spl/u-boot-spl.bin idbloader.img
	cp u-boot.itb ${OUT}/u-boot/
	cp idbloader.img ${OUT}/u-boot/
	cp ../rkbin/bin/rk35/rk3588_spl_loader_v1.08.111.bin ${OUT}/u-boot
	generate_spi_image
fi
```

ophub, 2023-04-14:
你发的u-boot文件，和我常见的文件大小偏差较大，我看不懂你这2个文件是怎么dd给镜像，跳过大小怎么设置？你有说明没？

gwx11111111, 2023-04-14:
hi,ophun
改一下我哪个uboot仓库有个问题
[image]
lib/avb/libavb/avb_slot_verify.c
把==NLLJ加上试试

gwx11111111, 2023-04-14:
=NULL加上

gwx11111111, 2023-04-14:
hi,ophub
  我有boot.img可以提前dtb吗？或者直接用编译好的dtb不行吗

ophub, 2023-04-14:
编译好了

gwx11111111, 2023-04-14:
HI,ophub
  厉害；现在还差什么文件呢？还是都有了

ophub, 2023-04-14:
https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_lede_04.14.0042
现在还差最后你测试下。

ophub, 2023-04-14:
用我上面修改的build脚本编译，完成后的u-boot在这，armbian和openwrt通用，所以2个仓库共用的，没重复保存。
https://github.com/ophub/amlogic-s9xxx-armbian/tree/main/build-armbian/u-boot/rockchip/ipc-r

gwx11111111, 2023-04-14:
有自带emmc，网卡有2个2.5g的，4个·1G的；
现在测试到还是不能开机呢？
=> reboot
DDR Version V1.08 20220617
LPDDR4X, 1560MHz
channel[0] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[1] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[2] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[3] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
Manufacturer ID:0x6 
change to F1: 528MHz
change to F2: 1068MHz
change to F3: 1560MHz
change to F0: 1560MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09-gc060f28d70-220414 #zyf (Apr 18 2022 - 18:13:34)
Failed to set cpub01
Failed to set cpub23
unknown raw ID phN
unrecognized JEDEC id bytes: 00, 00, 00
Trying to boot from MMC2
spl: partition error
Trying fit image at 0x4000 sector
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(ce2098067b...) + OK
## Checking uboot 0x00200000 ... sha256(e2fef55a2b...) + OK
## Checking fdt 0x00339a70 ... sha256(c07f4a4d71...) + OK
## Checking atf-2 0x000f0000 ... sha256(ebc45c531e...) + OK
## Checking atf-3 0xff100000 ... sha256(9ded9f3bb5...) + OK
## Checking optee 0x08400000 ... sha256(fde0860845...) + OK
Jumping to U-Boot(0x00200000) via ARM Trusted Firmware(0x00040000)
Total: 414.926 ms

INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-444-g1450d21e8:derrick.huang
NOTICE:  BL31: Built : 16:25:50, Oct 12 2022
INFO:    spec: 0x1
INFO:    ext 32k is not valid
INFO:    ddr: stride-en 4CH
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    valid_cpu_msk=0xff bcore0_rst = 0x0, bcore1_rst = 0x0
INFO:    system boots from cpu-hwid-0
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
INFO:    dfs DDR fsp_params[0].freq_mhz= 1560MHz
INFO:    dfs DDR fsp_params[1].freq_mhz= 528MHz
INFO:    dfs DDR fsp_params[2].freq_mhz= 1068MHz
INFO:    dfs DDR fsp_params[3].freq_mhz= 1560MHz
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
INFO:    BL31: Initializing BL32
INFO:    hdmirx_handler: dma not on, ret
I/TC: 
I/TC: OP-TEE version: 3.13.0-652-g4542e1efd #derrick.huang (gcc version 10.2.1 20201103 (GNU Toolchain for the A-profile Architecture 10.2-2020.11 (arm-10.16))) #5 2022年 09月 20日 星期二 09:41:09 CST aarch64
I/TC: Primary CPU initializing
I/TC: Primary CPU switching to normal world boot
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

U-Boot 2017.09-g48922e8182-dirty #smith (Apr 14 2023 - 04:45:18 +0000)

Model: Rockchip RK3588 Evaluation Board
PreSerial: 2, raw, 0xfeb50000
DRAM:  7.7 GiB
Sysmem: init
Relocation Offset: eda3d000
Relocation fdt: eb9fa738 - eb9fecc8
CR: M/C/I
Using default environment

no mmc device at slot 1
mmc@fe2c0000: 1 (SD), mmc@fe2e0000: 0
Bootdev(atags): mmc 1
MMC1: Legacy, 52Mhz
PartType: EFI
DM: v2
No misc partition
boot mode: None
FIT: No boot partition
No resource partition
No resource partition
No resource partition
Failed to load DTB, ret=-19
No find valid DTB, ret=-22
Failed to get kernel dtb, ret=-22
Model: Rockchip RK3588 Evaluation Board
rockchip_set_serialno: could not find efuse/otp device
CLK: (sync kernel. arm: enter 1008000 KHz, init 1008000 KHz, kernel 0N/A)
  b0pll 24000 KHz
  b1pll 24000 KHz
  lpll 24000 KHz
  v0pll 24000 KHz
  aupll 24000 KHz
  cpll 1500000 KHz
  gpll 1188000 KHz
  npll 24000 KHz
  ppll 100000 KHz
  aclk_center_root 702000 KHz
  pclk_center_root 100000 KHz
  hclk_center_root 396000 KHz
  aclk_center_low_root 500000 KHz
  aclk_top_root 750000 KHz
  pclk_top_root 100000 KHz
  aclk_low_top_root 396000 KHz
No misc partition
Net:   No ethernet found.
Hit key to stop autoboot('CTRL+C'):  0 
Could not find misc partition
ANDROID: reboot reason: "(none)"
optee api revision: 2.0
TEEC: Waring: Could not find security partition
Not AVB images, AVB skip
android_image_load_by_partname: Can't find part: boot
Android image load failed
Android boot failed, error -1.
## Booting FIT Image FIT: No boot partition
FIT: No fit blob
FIT: No FIT image
Unknown command 'bootrkp' - try 'help'
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
Found U-Boot script /boot.scr
3252 bytes read in 8 ms (396.5 KiB/s)
## Executing script at 00500000
Unknown command 'B' - try 'help'
SCRIPT FAILED: continuing...
switch to partitions #0, OK
mmc0(part 0) is current device
Failed to mount ext2 filesystem...
** Unrecognized filesystem type **

Device 2: unrecognized JEDEC id bytes: 00, 00, 00
unknown device

Device 1: unknown raw ID 0 0 0

Device 0: unknown device
starting USB...
No working controllers found
USB is stopped. Please issue 'usb start' first.
starting USB...
No working controllers found
No ethernet found.
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-rockchip
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default
No ethernet found.
Config file not found
starting USB...
No working controllers found
No ethernet found.
=>

gwx11111111, 2023-04-14:
机器也自带emmc的

gwx11111111, 2023-04-14:
HI,ophub
是不是内核不能用5.10.153呢？rk3588好像只有5.10.110的吧！可以用5.10.110来编译一下吗

ophub, 2023-04-14:
我删了153了，现在用110打包个试试，但是这个应该没影响，我rock5b（rk3588）用的153也正常使用

gwx11111111, 2023-04-14:
这样啊

gwx11111111, 2023-04-14:
那sd卡启动的话，需要
&sdmmc {
        status = "okay";
};这个sd卡的配置的吗？
或者usb怎么启动呢

ophub, 2023-04-14:
下新的测试

OpenWrt：
https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_lede_04.14.0249

Armbian：
https://github.com/ophub/amlogic-s9xxx-armbian/releases/tag/Armbian_bullseye_04.14.0244

gwx11111111, 2023-04-14:
好的，谢谢

gwx11111111, 2023-04-14:
HI ophub
   感觉是uboot要加入一些启动的引导才可以

ophub, 2023-04-14:
https://github.com/gwx11111111/uboot_ipc/tree/beelink_RK3358_rk12/arch/arm/dts

你这里缺你设备的dts文件，我放进去 rk3588-nvr-demo-v10-android.dts 相关的几个编译报错。

[image]

gwx11111111, 2023-04-14:
这个就是设备uboot下的./arch/arm/dts/rk3588-evb.dts dts啊或者想的话同名的话直接copy rk3588-evb.dts dts为rk3588-nvr-demo-v10-android.dts 也可以了

gwx11111111, 2023-04-14:
HI，ophub
     我这边拉不下来(https://github.com/radxa/build.git)这个仓库， 你那边能不能用 https://github.com/radxa/u-boot.git
这个仓库然后用这个rk3588_defconfig配置文件编译就是我们板子的配置了，看看是不是我哪个是安卓的uboot不行呢

gwx11111111, 2023-04-14:
uboot的dts在
[image]
这里指定的

ophub, 2023-04-14:
https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_lede_04.14.0446

你再试试这个。我用你的源码编译也没报错，怎么不能用呢？

[image]

dts 的 Makefile 我也加了，编译很顺利。

[image]

gwx11111111, 2023-04-14:
HI，ophub
我看了一下https://github.com/radxa/u-boot.git这个仓库的提交，
它里面加了很多关于引到的提交，但是我对比了安卓uboot，安卓
uboot没有的
像下面的：
[image]

gwx11111111, 2023-04-14:
HI,ophub
DDR Version V1.09 20220721
LPDDR4X, 1584MHz
channel[0] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[1] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[2] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[3] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
Manufacturer ID:0x6
change to F1: 528MHz
change to F2: 1068MHz
change to F3: 1560MHz
change to F0: 1584MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09-g08bbf419fb-dirty #zrb (Apr 06 2023 - 20:06:03)
Failed to set cpub01
Failed to set cpub23
unknown raw ID phN
unrecognized JEDEC id bytes: 00, 00, 00
Trying to boot from MMC2
spl: partition error
Trying fit image at 0x4000 sector
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(ce2098067b...) + OK
## Checking uboot 0x00200000 ... sha256(f1748af09d...) + OK
## Checking fdt 0x00339a70 ... sha256(c07f4a4d71...) + OK
## Checking atf-2 0x000f0000 ... sha256(ebc45c531e...) + OK
## Checking atf-3 0xff100000 ... sha256(9ded9f3bb5...) + OK
## Checking optee 0x08400000 ... sha256(fde0860845...) + OK
Jumping to U-Boot(0x00200000) via ARM Trusted Firmware(0x00040000)
Total: 419.406 ms

INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-444-g1450d21e8:derrick.huang
NOTICE:  BL31: Built : 16:25:50, Oct 12 2022
INFO:    spec: 0x1
INFO:    ext 32k is not valid
INFO:    ddr: stride-en 4CH
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    valid_cpu_msk=0xff bcore0_rst = 0x0, bcore1_rst = 0x0
INFO:    system boots from cpu-hwid-4
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
INFO:    dfs DDR fsp_params[0].freq_mhz= 1584MHz
INFO:    dfs DDR fsp_params[1].freq_mhz= 528MHz
INFO:    dfs DDR fsp_params[2].freq_mhz= 1068MHz
INFO:    dfs DDR fsp_params[3].freq_mhz= 1560MHz
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
INFO:    BL31: Initializing BL32
INFO:    hdmirx_handler: dma not on, ret
I/TC: 
I/TC: OP-TEE version: 3.13.0-652-g4542e1efd #derrick.huang (gcc version 10.2.1 20201103 (GNU Toolchain for the A-profile Architecture 10.2-2020.11 (arm-10.16))) #5 2022年 09月 20日 星期二 09:41:09 CST aarch64
I/TC: Primary CPU initializing
I/TC: Primary CPU switching to normal world boot
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

U-Boot 2017.09-g48922e8182-dirty #smith (Apr 14 2023 - 08:32:15 +0000)

Model: Rockchip RK3588 Evaluation Board
PreSerial: 2, raw, 0xfeb50000
DRAM:  7.7 GiB
Sysmem: init
Relocation Offset: eda3d000
Relocation fdt: eb9fa738 - eb9fecc8
CR: M/C/I
Using default environment

[... 146 lines cut ...]
        [< 0021764c >]
        [< 00217428 >]
        [< 00217948 >]
        [< 0021705c >]
        [< 0022d82c >]
        [< 0022e2a0 >]
        [< 00217794 >]
        [< 00217948 >]
        [< 0021705c >]
        [< 0022d790 >]
        [< 00217fe8 >]
        [< 00215e7c >]
        [< 00218680 >]
        [< 002a0bb4 >]
        [
[image]
< 002188b4 >]
        [< 00201e18 >]

Copy info from "Call trace..." to a file(eg. dump.txt), and run
command in your U-Boot project: ./scripts/stacktrace.sh dump.txt 

Resetting CPU ...
根据这里看
还像还是安卓的uboot不行

gwx11111111, 2023-04-14:
Hi,ophub
测试了上面最新的，有以上的报错

ophub, 2023-04-14:
[image]
[image]
[image]
[image]

https://github.com/radxa/build/blob/debian/board_configs.sh
添加了选项

```
	"rk3588-ipc")
		DEFCONFIG=rockchip_linux_defconfig
		UBOOT_DEFCONFIG=ipc-rk3588_defconfig
		DTB=rk3588-evb.dtb
		export ARCH=arm64
		export CROSS_COMPILE=aarch64-linux-gnu-
		CHIP="rk3588"
		;;
```

https://github.com/radxa/build/blob/debian/mk-uboot.sh
根据你的rkbin源码修改了版本和文件

```
elif [ "${CHIP}" == "rk3588s" ] || [ "${CHIP}" == "rk3588" ]; then
	make ${UBOOT_DEFCONFIG}
	make BL31=../rkbin/bin/rk35/rk3588_bl31_v1.30.elf spl/u-boot-spl.bin u-boot.dtb u-boot.itb
	./tools/mkimage -n rk3588 -T rksd -d ../rkbin/bin/rk35/rk3588_ddr_lp4_2112MHz_lp5_2736MHz_v1.08.bin:spl/u-boot-spl.bin idbloader.img
	cp u-boot.itb ${OUT}/u-boot/
	cp idbloader.img ${OUT}/u-boot/
	cp ../rkbin/bin/rk35/rk3588_spl_* ${OUT}/u-boot
	generate_spi_image
fi
```

gwx11111111, 2023-04-14:
HI，ophub
    上面的测试固件就是https://github.com/radxa/u-boot.git的吗？还是radxa的uboot还在编译呢

ophub, 2023-04-14:
https://github.com/ophub/amlogic-s9xxx-armbian/commit/480a0087b9893cce0d1eb7abf7de2b5b1647a721
https://github.com/radxa/u-boot.git 的刚更新，正在打包，5分钟后可以测试固件。
https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/4698564136

https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/armbian-files/platform-files/rockchip/bootfs/dtb/rockchip/rk3588-nvr-demo-v10-android.dtb

你这个dtb是5.10.100编译的吗？

gwx11111111, 2023-04-14:
是安卓内核5.10.110编译的

ophub, 2023-04-14:
你把你的 rk3588-nvr-demo-v10-android.dts 放在 https://github.com/unifreq/linux-5.10.y-rk35xx 这个里编译下看看，5b的 u-boot和kernel用这个现在几个rk3588设备都可以正常运行.

gwx11111111, 2023-04-14:
HI,ophub
     如果改内核的dtb不是得改https://github.com/unifreq/linux-5.10.y-rk35xx.git这个仓库的dts吗？
为啥直接放到[files/rockchip/bootfs/dtb/rockchip/rk3588-nvr-demo-v10-android.dtb](https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/armbian-files/platform-files/rockchip/bootfs/dtb/rockchip/rk3588-nvr-demo-v10-android.dtb)这也可以呢

ophub, 2023-04-14:
https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_lede_04.14.0558

用radxa的u-boot重新打包的op，你试试

gwx11111111, 2023-04-14:
好的

gwx11111111, 2023-04-14:
谢谢了

ophub, 2023-04-14:
你直接放他就不编译，能不能用忽略了。
你放到kernel源码下编译dts，要保障你的这个dtb是可以和u-boot正常配合使用的。
现在u-boot顺利编译，用你的和radxa的都可以编译完成，如果你使用的是rk3588-evb的话现在编译的u-boot就可以使用，过程在上面的帖子都说明了。可能是dtb的问题了。

ophub, 2023-04-14:
https://github.com/ophub/amlogic-s9xxx-armbian/tree/main/build-armbian/armbian-files/platform-files/rockchip/bootfs/dtb/rockchip

这里是内核tar解压的文件，在制作op时从内核仓库下载最新文件，解压并覆盖上面的同名文件。现在内核tar包里没有你的这个dtb，所以放解压包里，制作op时没有同名文件覆盖，就保留下来了，以后添加到内核里以后，这个目录里面的就没用了，每次都会被覆盖了

https://github.com/ophub/kernel/releases/tag/kernel_rk3588

gwx11111111, 2023-04-14:
HI,ophub
    好的，了解
现在最新的tf卡启动可以开机了，但是一直
[    2.504005] es8311 3-0018: ASoC: error at soc_component_read_no_lock on es8311.3-0018: -6
[    2.504152] es8311 3-0018: ASoC: error at soc_component_read_no_lock on es8311.3-0018: -6
[    2.504299] es8311 3-0018: ASoC: error at soc_component_read_no_lock on es8311.3-0018: -6
[    2.504447] es8311 3-0018: ASoC: error at soc_component_write_no_lock on es8311.3-0018: -6
[    2.504597] es8311 3-0018: ASoC: error at soc_component_read_no_lock on es8311.3-0018: -6
[    2.504744] es8311 3-0018: ASoC: error at soc_component_write_no_lock on es8311.3-0018: -6
[    2.504892] es8311 3-0018: ASoC: error at soc_component_write_no_lock on es8311.3-0018: -6
[    2.505040] es8311 3-0018: ASoC: error at soc_component_write_no_lock on es8311.3-0018: -6
[    2.505884] es8311 3-0018: ASoC: error at soc_component_read_no_lock on es8311.3-0018: -6
[    2.506038] es8311 3-0018: ASoC: error at soc_component_read_no_lock on es8311.3-0018: -6
[    2.509218] ALSA device list:
[    2.509228]   #0: rockchip,dp1
[    2.509234]   #1: rockchip,hdmi0
[    2.509238]   #2: rockchip,es8311
[    2.513022] Freeing unused kernel memory: 7488K
[    2.521967] Run /init as init process
[    2.540201] vendor storage:20190527 ret = 0
[    2.563755] usb 2-1: New USB device found, idVendor=05e3, idProduct=0608, bcdDevice=85.36
[    2.563798] usb 2-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    2.563814] usb 2-1: Product: USB2.0 Hub
[    2.564655] hub 2-1:1.0: USB hub found
[    2.564993] hub 2-1:1.0: 4 ports detected
[    2.715420] mali fb000000.gpu: Kernel DDK version g15p0-01eac0
[    2.718011] mali fb000000.gpu: bin=0
[    2.718207] mali fb000000.gpu: leakage=21
[    2.718269] debugfs: Directory 'fb000000.gpu-mali' with parent 'vdd_gpu_s0' already present!
[    2.725246] mali fb000000.gpu: pvtm=866
[    2.726339] mali fb000000.gpu: pvtm-volt-sel=3
[    2.726839] mali fb000000.gpu: avs=0
[    2.726927] W : [File] : drivers/gpu/arm/bifrost/platform/rk/mali_kbase_config_rk.c; [Line] : 136; [Func] : kbase_platform_rk_init(); power-off-delay-ms not available.
[    2.738775] mali fb000000.gpu: GPU hardware issue table may need updating:
[    2.738775] r0p0 status 5 is unknown; treating as r0p0 status 0
[    2.738800] mali fb000000.gpu: GPU identified as 0x7 arch 10.8.6 r0p0 status 0
[    2.738845] mali fb000000.gpu: No priority control manager is configured
[    2.738942] mali fb000000.gpu: No memory group manager is configured
[    2.738964] mali fb000000.gpu: Protected memory allocator not available
[    2.739429] mali fb000000.gpu: Capping CSF_FIRMWARE_TIMEOUT to CSF_FIRMWARE_PING_TIMEOUT
[    2.740236] mali fb000000.gpu: Couldn't find power_model DT node matching 'arm,mali-simple-power-model'
[    2.740257] mali fb000000.gpu: Error -22, no DT entry: mali-simple-power-model.static-coefficient = 1*[0]
[    2.740484] mali fb000000.gpu: Error -22, no DT entry: mali-simple-power-model.dynamic-coefficient = 1*[0]
[    2.740649] mali fb000000.gpu: Error -22, no DT entry: mali-simple-power-model.ts = 4*[0]
[    2.740814] mali fb000000.gpu: Error -22, no DT entry: mali-simple-power-model.thermal-zone = ''
[    2.743178] mali fb000000.gpu: Using configured power model mali-lodx-power-model, and fallback mali-simple-power-model
[    2.743386] mali fb000000.gpu: l=10000 h=85000 hyst=5000 l_limit=0 h_limit=800000000 h_table=0
[    2.762016] mali fb000000.gpu: Probed as mali0
卡在这里

[332 more replies]

closed 2023-05-19 by ophub