ophub/amlogic-s9xxx-openwrt #416: Enable Wi-Fi / LAN Tanix-TX6 / MX10 Pro

netrixken, 2023-04-13:
Standard chat template, no routine, no chat.
标准聊天模板，无套路不聊天。

**Device Information | 设备信息**
 - SOC: [e.g. S095x3] h6
 - Model [e.g. HK1] tanix-tx6

**Armbian Version | 系统版本**
 - Kernel Version: [e.g. 5.15.100] 6.1.23-ophub
 - Release: [e.g. lede] OpenWrt 22.03.3 r20028-43d71ad93e / LuCI openwrt-22.03 branch git-22.361.69894-438c598

**Describe the bug | 问题描述**
Describe the problem in detail and attach screenshots if necessary.
详细描述问题，并在必要时附上屏幕截图。

Currently I am using Tanix TX6 on MX10 Pro which has identical hardware. Everything is works, except LAN & WiFi (XR819).

Is there anything I can do to enable WiFi or LAN? Maybe how-to build working xradio? (I read somewhere it's much stable now)

Setting up OpenWRT without working LAN is very tiring. Btw, I tried using https://github.com/fifteenhex/xradio with no success. Here's the log:

```
Thu Apr 13 12:50:33 2023 kern.warn kernel: [   44.973980] xradio_wlan: loading out-of-tree module taints kernel.
Thu Apr 13 12:50:33 2023 kern.warn kernel: [   44.987226] ------------[ cut here ]------------
Thu Apr 13 12:50:33 2023 kern.warn kernel: [   44.994002] sunxi-mmc 4021000.mmc: DMA addr 0x000000404d313830+8 overflow (mask ffffffff, bus limit 0).
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.005578] WARNING: CPU: 1 PID: 2197 at kernel/dma/direct.h:103 dma_direct_map_sg+0x2a8/0x2cc
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.016369] Modules linked in: xradio_wlan(O+) pppoe ppp_async nft_fib_inet nf_flow_table_inet ath9k_htc ath9k_common rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib rt2500usb pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables mt76x0u mt76x0_common brcmfmac ath9k_hw ath10k_usb ath10k_sdio ath10k_core slhc rtlwifi rtl8150 rt2x00usb rt2x00lib r8188eu(C) nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_nat nf_log_syslog nf_flow_table mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt7663u mt7663_usb_sdio_common mt7615_common mt7601u mt76_usb mt76_connac_lib mt76 meson_gxbb_wdt mac80211 libarc4 brcmutil ax88179_178a ath 88x2cs cfg80211 rfkill r8153_ecm cdc_ether usbnet r8152 dw_hdmi_cec ohci_platform ohci_hcd panfrost gpu_sched drm_shmem_helper
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.118496] CPU: 1 PID: 2197 Comm: insmod Tainted: G         C O       6.1.23-ophub #1
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.129135] Hardware name: Tanix TX6 (DT)
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.135822] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.145490] pc : dma_direct_map_sg+0x2a8/0x2cc
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.152625] lr : dma_direct_map_sg+0x2a8/0x2cc
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.159697] sp : ffffffc00d3133b0
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.165642] x29: ffffffc00d3133b0 x28: 000000404d313830 x27: ffffffc00d3135b8
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.175486] x26: 0000000000000000 x25: ffffff8002644010 x24: 0000000000000002
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.185366] x23: 0000000000000000 x22: fffffffe00000000 x21: ffffffc0097748a0
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.195196] x20: 0000000000000001 x19: 0000000000000000 x18: ffffffffffffffff
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.204975] x17: 6d2820776f6c6672 x16: 65766f20382b3033 x15: 3833313364343034
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.214797] x14: 3030303030307830 x13: ffffffc009d1cdf8 x12: 00000000000005be
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.224652] x11: 00000000000001ea x10: ffffffc009dccdf8 x9 : ffffffc009d1cdf8
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.234534] x8 : 00000000ffffdfff x7 : ffffffc009dccdf8 x6 : 0000000000000001
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.244432] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.254316] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff8003c1f000
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.264185] Call trace:
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.269333]  dma_direct_map_sg+0x2a8/0x2cc
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.276171]  __dma_map_sg_attrs+0x28/0x9c
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.282936]  dma_map_sg_attrs+0x10/0x24
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.289496]  sunxi_mmc_request+0x70/0x4b0
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.296201]  __mmc_start_request+0x68/0x140
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.303057]  mmc_start_request+0x94/0xc0
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.309611]  mmc_wait_for_req+0x70/0x100
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.316116]  mmc_io_rw_extended+0x1d4/0x2cc
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.322881]  sdio_io_rw_ext_helper+0x194/0x240
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.329910]  sdio_memcpy_fromio+0x20/0x30
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.336492]  sdio_data_read+0x3c/0x80 [xradio_wlan]
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.343991]  __xradio_read+0x2c/0x70 [xradio_wlan]
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.351381]  xradio_reg_read+0x40/0x6c [xradio_wlan]
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.358946]  xradio_load_firmware+0x38/0x1430 [xradio_wlan]
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.367132]  xradio_core_init+0x13c/0x45c [xradio_wlan]
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.374978]  sdio_probe+0x118/0x224 [xradio_wlan]
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.382302]  sdio_bus_probe+0x134/0x1a0
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.388698]  really_probe+0xc0/0x3dc
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.394806]  __driver_probe_device+0x7c/0x190
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.401681]  driver_probe_device+0x40/0xe0
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.408276]  __driver_attach+0xf4/0x200
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.414558]  bus_for_each_dev+0x70/0xd0
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.420787]  driver_attach+0x24/0x30
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.426677]  bus_add_driver+0x17c/0x240
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.432752]  driver_register+0x78/0x130
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.438747]  sdio_register_driver+0x28/0x3c
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.445018]  xradio_sdio_register+0x18/0x24 [xradio_wlan]
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.452476]  xradio_core_entry+0x10/0x1000 [xradio_wlan]
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.459775]  do_one_initcall+0x50/0x1d0
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.465491]  do_init_module+0x48/0x1d0
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.471040]  load_module+0x19a0/0x1de0
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.476552]  __do_sys_finit_module+0xac/0x130
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.482652]  __arm64_sys_finit_module+0x20/0x30
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.488901]  invoke_syscall+0x48/0x114
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.494340]  el0_svc_common.constprop.0+0x50/0x18c
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.500848]  do_el0_svc+0x2c/0xc0
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.505882]  el0_svc+0x2c/0xbc
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.510632]  el0t_64_sync_handler+0xf4/0x120
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.516585]  el0t_64_sync+0x19c/0x1a0
Thu Apr 13 12:50:34 2023 kern.warn kernel: [   45.521918] ---[ end trace 0000000000000000 ]---
Thu Apr 13 12:50:34 2023 kern.err kernel: [   45.528276] sunxi-mmc 4021000.mmc: dma_map_sg failed
Thu Apr 13 12:50:34 2023 kern.err kernel: [   45.534940] sunxi-mmc 4021000.mmc: map DMA failed
Thu Apr 13 12:50:34 2023 kern.err kernel: [   45.541392] xradio_wlan mmc2:0001:1: bh thread exiting
```

netrixken, 2023-04-15:
Found linux kernel patch for enable LAN & WiFi

https://github.com/ktkd/linux-5.7-h6-mainline

ophub, 2023-04-15:
[image]
[image]
[image]
[image]
[image]

In the 5.10, 5.15, and 6.1 kernels, there are a lot of places that cannot be patched. The version difference is too large, and the patch is not applicable.

closed 2023-04-15 by ophub